toitlang/toit #9: Build issue on macOS Monterey with Xcode 13.1 due to mbedtls

kendfinger, 2021-11-22:
It appears that with the default instructions for Linux, as applied to macOS, the build fails due to mbedtls in the recommended branch and new clang warnings involving potentially incorrect string concatenation.

**Environment**:

OS: macOS 12.0.1 (21A559) Darwin 21.1.0
Hardware: Apple M1 Max
Architecture: arm64
Xcode: 13.1 (13A1030d)

**Clang**:

```
Apple clang version 13.0.0 (clang-1300.0.29.3)
Target: arm64-apple-darwin21.1.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

esp-idf:  patch-head-4.3-3 (6f21fa91898d1b409e5de064abd75eea5e3a3464) from https://github.com/toitware/esp-idf

**Workaround**: Removing -Werror in TOIT_GENERIC_FLAGS in CMakeLists.txt does fix it, so this appears to just be a combination of -Wall -Werror and some overly cautious new warnings in Clang.

```
FAILED: bin/mbedtls/library/CMakeFiles/mbedcrypto.dir/ripemd160.c.o
CCACHE_BASEDIR=/Users/kendfinger/Developer/External/Toit/toit /opt/homebrew/bin/ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -I/Users/kendfinger/Developer/External/Toit/esp-idf/components/mbedtls/mbedtls/include '-fdebug-prefix-map=/Users/kendfinger/Developer/External/Toit/toit=.' -Wall -Werror -ffunction-sections -fdata-sections -Wno-unused-command-line-argument  -DMBEDTLS_SSL_IN_CONTENT_LEN=4608 -DMBEDTLS_SSL_OUT_CONTENT_LEN=3700 -DMBEDTLS_PLATFORM_MEMORY=1 -Wall -Wextra -W -Wdeclaration-after-statement -Wwrite-strings -Wpointer-arith -Wimplicit-fallthrough -Wshadow -Wmissing-declarations -Wmissing-prototypes -Wdocumentation -Wno-documentation-deprecated-sync -Wunreachable-code -O2 -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk -MD -MT bin/mbedtls/library/CMakeFiles/mbedcrypto.dir/ripemd160.c.o -MF bin/mbedtls/library/CMakeFiles/mbedcrypto.dir/ripemd160.c.o.d -o bin/mbedtls/library/CMakeFiles/mbedcrypto.dir/ripemd160.c.o -c /Users/kendfinger/Developer/External/Toit/esp-idf/components/mbedtls/mbedtls/library/ripemd160.c
/Users/kendfinger/Developer/External/Toit/esp-idf/components/mbedtls/mbedtls/library/ripemd160.c:518:7: error: suspicious concatenation of string literals in an array initialization; did you mean to separate the elements with a comma? [-Werror,-Wstring-concatenation]
      "345678901234567890" },
      ^
/Users/kendfinger/Developer/External/Toit/esp-idf/components/mbedtls/mbedtls/library/ripemd160.c:517:7: note: place parentheses around the string literal to silence warning
    { "12345678901234567890123456789012345678901234567890123456789012"
      ^
1 error generated.
```

[build.log](https://github.com/toitlang/toit/files/7580523/build.log)

floitsch, 2021-11-22:
Ideally, there should be a way to disable `-Werror` for submodules as we don't have a (super) easy way to fix warnings there.

If I don't find a good way to do that it might be best to remove `-Werror`.

bolshoytoster, 2021-11-25:
This has been fixed in the official mbedtls repository (https://github.com/ARMmbed/mbedtls/pull/3588) but you could probably get away with just removing the `-Werror` from line 38 in the `CMakeLists.txt` file.

kasperl, 2021-12-13:
Fixed in 35c4e9e49087fcb661a6002c2e2314cfefaa64fc.

closed 2021-12-13 by kasperl