toitlang/toit #25: Driver for touch screen on M5Stack Core2

kasperl, 2021-11-26:
Porting existing drivers to Toit is a great place to start contributing to the Toit community. The M5Stack Core2 is a fantastic device. It comes with MIT licensed support for the touch support for the screen, but that support isn't yet available for Toit users:

https://github.com/m5stack/M5Core2/blob/master/src/M5Touch.h

It appears to be based on the FocalTech FT6336 chip and it uses I2C to communicate. It would be great to have a Toit package that makes the functionality available to all Toit users.

Here is the [source code of another driver package](https://github.com/imliubo/mpu6886-toit) that is useful on the M5Core and here is brief explanation on 
[how to start writing an I2C driver in Toit](https://github.com/toitlang/toit/discussions/22).

Happy hacking!

kasperl, 2021-11-26:
Mentioned your work on the MPU6886 here, @imliubo. Thanks again!

imliubo, 2021-11-26:
Thanks,has anyone start for touch upport? Maybe I can try it later, but I'm still stuck in other work...

kasperl, 2021-11-26:
Not to my knowledge!

imliubo, 2021-11-29:
A very simple driver for FT63xx capacitive touch chip.
[Link](https://github.com/imliubo/ft63xx-toit)

kasperl, 2021-12-01:
Fixed by @imliubo here: https://pkg.toit.io/package/github.com%2Fimliubo%2Fft63xx-toit.

Fantastic work!

closed 2021-12-01 by kasperl