toitlang/toit #27: Driver for Modbus TCP/IP

kasperl, 2021-11-26:
[Modbus TCP/IP](https://www.adafruit.com/product/1782) is a data communications protocol used for connecting industrial electronic devices. There is no Toit driver package available for it yet. If you know how the protocol works or if you're interesting in figure it out, we could use your help to build Toit support for it.

Porting existing drivers to Toit is a great place to start contributing to the Toit community. Here is the [source code of another driver package](https://github.com/toitware/toit-lsm303dlhc) for inspiration. Working on Modbus will involve a lot of bit-fiddling, but at least you don't have to be familiar with I2C to get involved.

gregger2k, 2021-11-28:
Don't leave out Modbus Serial!
Not everything has TCP/IP

kendfinger, 2021-11-30:
I would love to take this on in my free time! Modbus has a bit of a special place in my heart as an old protocol that is still deployed widely.

@gregger2k I will be more than happy to work on Modbus RTU as well, in fact to implement TCP/IP you are are just wrapping Modbus RTU in a header. Also common is Modbus RTU over TCP (RTU encapsulation), which is often used when a serial to ethernet converter is used.

kasperl, 2022-02-17:
First version of this has been published: https://github.com/toitlang/pkg-modbus. Further contributions are very welcome, @kendfinger!

closed 2022-02-17 by kasperl