toitlang/toit #29: Driver for Sensirion SCD30

kasperl, 2021-11-26:
The [Sensirion SCD30](https://www.sensirion.com/en/environmental-sensors/carbon-dioxide-sensors/carbon-dioxide-sensors-scd30/) is a sensor module for HVAC and indoor air quality applications. There is no Toit driver package available for it yet. If you have an SCD30 sensor, we could use your help to build Toit support for it.

Porting existing drivers to Toit is a great place to start contributing to the Toit community. Here is the [source code of another driver package](https://github.com/toitware/toit-lsm303dlhc) for inspiration and here is brief explanation on [how to start writing an I2C driver in Toit](https://github.com/toitlang/toit/discussions/22).

juergenspecht, 2021-11-26:
I have 4 of these sensors lying around, so happy to help with the testing. However, I am still a Toit beginner, so not able to actually write this driver.

floitsch, 2021-12-24:
A driver for the scd30 now exists: https://pkg.toit.io/package/github.com%2Fqvisten999%2Fscd30

closed 2021-12-24 by floitsch