toitlang/toit #32: Is a BLE Server on the map?

justind000, 2021-11-26:
Just writing to see if a BLE server implementation is being worked on. Specifically, what I'd like to see is turning the ESP32 into a server that other devices connect to, that creates characteristics and updates their values. 

I have taken a look through [ble.toit ](https://github.com/toitlang/toit/blob/master/lib/ble.toit) and it looks like some of the primitives are already there. 

- [x] Device
- [x] Characteristic
- [ ] Security
- [ ] Server
- [ ] Service

Toit is using the NimBLE implementation, if I'm looking at [this ](https://github.com/toitlang/toit/blob/40020a46a3068912bea2e344b893ceb4c1a3eb29/toolchains/esp32/sdkconfig#L248)correctly?

kasperl, 2021-11-27:
Yes, you're right. We are using NimBLE. We are not actively working on adding the last parts of the BLE implementation right now (as in the coming week), but we'd love to explore this with you.

For reference, the [resources/ble_esp32.cc](https://github.com/toitlang/toit/blob/master/src/resources/ble_esp32.cc) and [event_sources/ble_esp32.cc](https://github.com/toitlang/toit/blob/master/src/event_sources/ble_esp32.cc) files contain the implementation of the primitives used by [ble.toit](https://github.com/toitlang/toit/blob/master/lib/ble.toit).

kasperl, 2021-11-27:
I've taken the liberty of turning your question into a feature request :)

justind000, 2021-11-27:
I'm interested in the platform and am currently writing a few I2C drivers to get familiar with it. 

A BLE server isn't on my short list either, but I have found it very useful to have for provisioning and interacting with the device through an app ([web BLE](https://googlechrome.github.io/samples/web-bluetooth/) maybe since you are already using web serial). 

If I start playing around with it and make some progress, I will get in touch here.

kasperl, 2021-11-27:
Thanks, @justind000!

mikkeldamsgaard, 2022-01-28:
I am currently giving the ble server/peripheral some attention. The idea is to let the user build a service descriptor with toit objects to be passed to the server constructor. The library would then enable advertising the services. It will contain some sensible defaults for attribute types and allow callbacks to executed in user code when an attribute is written to, enable reading an attribute and provide a default mechanism to handle notifications.

kasperl, 2022-02-28:
Mikkel's work just landed. Please take a look, @justind000!

kasperl, 2022-08-20:
With a good chunk of help from @mikkeldamsgaard, the BLE implementation is in a much better shape now. Do you know what we're still missing, @justind000?

mikkeldamsgaard, 2022-11-24:
Still missing security. The rest is done.

mikkeldamsgaard, 2023-03-07:
Security finally added in https://github.com/toitlang/toit/pull/1470. @kasperl please update when approved.

closed 2023-03-09 by floitsch