toitlang/toit #35: Ensure that standard libraries don't try to use System process

floitsch, 2021-12-01:
The OSS edition doesn't have any system process. However, some libraries still use RPC to request information (for example `device.hardware_id`).

These libraries need to be rewritten.

kasperl, 2022-02-28:
This has been fixed.

dumblob, 2023-06-27:
Could you please provide more details (link to source code?) how this got fixed?

Thank you!

kasperl, 2023-06-28:
Hi @dumblob! 

When we open sourced We had a few places where we would use a remote procedure call mechanism to implement library functionality. The library code assumed that we would be running in a context where a system process would implement the remote procedures, but we dropped that as part of open sourcing. We have later introduced the system process again.

To give you a sense for the kind of changes we made, you can see one such change here:

https://github.com/toitlang/toit/commit/281c67032b03548c8fcade2c5412872368c7f93b

dumblob, 2023-06-29:
Ok, thanks for the reference.

It seems still a bit foggy how the system process looks like nowadays and how it interacts with standard libraries (these presumably should not be allowed to do so with a few notable exceptions I suppose).

But it seems it is more of na issue on my side not having enough time to read all the relevant sources in this repo.