toitlang/toit #37: Make it possible to deep sleep

floitsch, 2021-12-01:
We need to expose the deep_sleep functionality in a nice way.

kasperl, 2021-12-01:
Adding a method like this somewhere in the core libraries, could be a good first step:

```
enter_deep_sleep time/Duration -> none:
  __deep_sleep__ time.in_ms
```

This allows you to enter deep sleep for 30 seconds simply by writing:

```
enter_deep_sleep (Duration --s=30)
```

justind000, 2021-12-02:
Sleeping for a duration and sleeping until a pin interrupt occurs are the two methods I use most commonly: sending measurements every x time-period and doing something when a button is pressed.

kasperl, 2021-12-03:
Thanks for the comment, @justind000. Would something like these two methods - `enable_external_wakeup` and `wakeup_cause` serve your needs?

```
/**
  Set the ESP32 to wake up from deep sleep if the GPIO pins in pin_mask matches the mode.
  if on_any_high==true, the ESP32 will wake up if any pin in the mask is high.
  if on_any_high==false, the ESP32 will wake up if all pins in the mask are low.
  Only the following GPIO pins can be used: 0,2,4,12-15,25-27,32-39.
*/
enable_external_wakeup pin_mask/int on_any_high/bool -> none:
  ...

// Enum constants from esp_sleep_source_t in esp-idf/components/esp32/include/esp_sleep.h.
WAKEUP_UNDEFINED ::= 0 // In case of deep sleep, reset was not caused by exit from deep sleep.
WAKEUP_ALL       ::= 1 // Not a wakeup cause, used to disable all wakeup sources with esp_sleep_disable_wakeup_source.
WAKEUP_EXT0      ::= 2 // Wakeup caused by external signal using RTC_IO.
WAKEUP_EXT1      ::= 3 // Wakeup caused by external signal using RTC_CNTL.
WAKEUP_TIMER     ::= 4 // Wakeup caused by timer.
WAKEUP_TOUCHPAD  ::= 5 // Wakeup caused by touchpad.
WAKEUP_ULP       ::= 6 // Wakeup caused by ULP program.
WAKEUP_GPIO      ::= 7 // Wakeup caused by GPIO (light sleep only).
WAKEUP_UART      ::= 8 // Wakeup caused by UART (light sleep only).

wakeup_cause -> int:
  ...
```

justind000, 2021-12-03:
Yes, it would. I make sensor devices, so for me, the typical application flow looks like this:

1. init system and check wakeup reason, if GPIO, that means a button was pressed by the user to do something (usually configuration related)
2. take and process sensor measurement
3. set GPIO interrupt (connected to a button) and deep sleep for x duration

kasperl, 2021-12-04:
Started working on this.

kasperl, 2021-12-04:
@justind000 You can check out the https://github.com/toitlang/toit/tree/deep-sleep branch and try the example:

```
import esp32

main:
  print "Ran for $(Duration --us=esp32.total_run_time) so far"
  print "Slept for $(Duration --us=esp32.total_deep_sleep_time) so far"
  esp32.deep_sleep (Duration --s=10)
```

This leads to this serial output:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:188
ho 0 tail 12 room 4
load:0x40078000,len:12180
load:0x40080400,len:2944
entry 0x400805c8
clearing RTC memory: RTC memory is in inconsistent state
[flash reg] address 0x3f42c000, size 0x00134000
Ran for 495.492ms so far
Slept for 0s so far
Entering deep sleep for 10000ms
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:188
ho 0 tail 12 room 4
load:0x40078000,len:12180
load:0x40080400,len:2944
entry 0x400805c8
[flash reg] address 0x3f42c000, size 0x00134000
Ran for 10.561596s so far
Slept for 10.048755s so far
Entering deep sleep for 10000ms
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:188
ho 0 tail 12 room 4
load:0x40078000,len:12180
load:0x40080400,len:2944
entry 0x400805c8
[flash reg] address 0x3f42c000, size 0x00134000
Ran for 20.651116s so far
Slept for 20.120397s so far
Entering deep sleep for 10000ms
```

PR #63 (2021-12-04): Add support for deep sleep on the ESP32.
This fixes #37.
commit 055cfc21 (2021-12-04): Add support for deep sleep on the ESP32. (#63)

* Add support for deep sleep on the ESP32.

This fixes #37.

* Update lib/esp32.toit

* Update lib/esp32.toit

* Update lib/esp32.toit

* Update lib/esp32.toit

* Move comment fixes.

* Tweak comments and improve example.
closed 2021-12-04 by kasperl