toitlang/toit #40: Add support for ESP32-S3

floitsch, 2021-12-01:


floitsch, 2022-01-31:
@erikcorry do you have some information?

un01s, 2022-02-02:
Start to build esp-idf examples with its built-in tinyUSB and load it to esp32s3 dev board. Will update as it goes.

erikcorry, 2022-02-03:
I merged a CL that removed Xtensa-specific things from cmpctmalloc.  I hope it now compiles on RISCV.

kasperl, 2022-08-20:
Hi @un01s! We've upgraded to ESP-IDF v4.4 which may (or may not) make this easier to make progress on.

nivoc, 2022-09-04:
@erikcorry But ESP32-S3 has no RISC Processor or? The RISC one is ESP32-C3 and that is Single Core too.

https://en.wikipedia.org/wiki/ESP32#ESP32-C3
https://en.wikipedia.org/wiki/ESP32#ESP32-S3

floitsch, 2022-09-16:
@nivoc: you are correct.

The biggest problem with the S3 is that Espressif only supports it with CMake. However, Toit still uses the old (deprecated) make approach.

The code should mostly work (except that some Pin constants would likely need to be adapted). There is one user of Toit who runs successfully on the S3 (with a few hacks to get the build system working).

kasperl, 2022-11-23:
S3 support has landed :)

closed 2022-11-23 by kasperl