toitlang/toit #43: Add RISC-V SiFive Unmatched platform support

dsobotta, 2021-12-01:
I've made some good progress on the SiFive Unmatched.  For anyone else looking to contribute, feel free to use [my fork](https://github.com/dsobotta/toit-riscv) as a starting point.  I'll issue a PR back into mainline once things have stabilized a bit.

dsobotta, 2021-12-01:
Currently blocked by [is_directory: Invalid argument](https://github.com/dsobotta/toit-riscv/issues/3) .  This is reproducible in a QEMU/riscv64 environment following my [build steps](https://github.com/dsobotta/toit-riscv).  Is this something you could take a look at @kasperl ?

kasperl, 2021-12-02:
Will take a look, @dsobotta!

dsobotta, 2021-12-04:
Made some good progress on cross-compiling.

I added a few entries to 'Makefile' (and added toolchains/riscv64.cmake) to get cross-compiling of Toit binaries from amd64 host -> riscv64 working.  It successfully generates build/riscv64/bin/{lib/, mbedtls, toitc, toitvm}.  These binaries have been validated on RISC-V hardware, and appear to run the examples with the same results as toitvm/toitc binaries generated directly from a RISC-V host.

There's one snag though--the toitvm_boot.snapshot is not successfully generated as a part of the cross-compile build, as the build scripts attempt to run the riscv64 ELF when compiling it.  Is there a cmake var I can override in toolchains/riscv64.cmake , or will I need to refactor some portion of the build system to break out that step?

Here's the [github diff ](https://github.com/dsobotta/toit-riscv/commit/ff6353ece167cfe617a119775d7a65a16eb45e7a?diff=unified)with the current Makefile changes.

dsobotta, 2021-12-26:
With the [makefile refactor](https://github.com/toitlang/toit/pull/197) merged, we now have end-to-end support for riscv64/linux targets and cross-compiling from other hosts.  I don't suspect that many will be deploying to ESP32 devices from riscv64 hosts, so closing this out.

closed 2021-12-26 by dsobotta