toitlang/toit #45: components/lwip/lwip issue on Toitware esp-idf fork

mikini, 2021-12-02:
There seems to be a problem with the registered hash of the components/lwip/lwip submodule in Toitware esp-idf fork.

Relevant excerpt when following README.md procedure:
```
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
...
Cloning into 'components/lwip/lwip'...
remote: Enumerating objects: 50148, done.
remote: Counting objects: 100% (182/182), done.
remote: Compressing objects: 100% (121/121), done.
remote: Total 50148 (delta 108), reused 110 (delta 59), pack-reused 49966
Receiving objects: 100% (501Cloning into 'components/lwip/lwip'...48/50148), 9.94 MiB | 2.56 MiB/s, done.
Resolving deltas: 100% (37849/37849), done.
Checking connectivity... done.
fatal: reference is not a tree: 02638065e72ea689eb0d3509b1960b31c515616e
...
Unable to checkout '02638065e72ea689eb0d3509b1960b31c515616e' in submodule path 'components/lwip/lwip'

```

Complete dump:
```
$ git clone https://github.com/toitlang/toit toit_new
Cloning into 'toit_new'...
remote: Enumerating objects: 1113, done.
remote: Counting objects: 100% (1113/1113), done.
remote: Compressing objects: 100% (879/879), done.
remote: Total 1113 (delta 307), reused 1005 (delta 210), pack-reused 0
Receiving objects: 100% (1113/1113), 4.91 MiB | 2.94 MiB/s, done.
Resolving deltas: 100% (307/307), done.
Checking connectivity... done.
$ cd toit_new/
$ git clone https://github.com/toitware/esp-idf.git
Cloning into 'esp-idf'...
remote: Enumerating objects: 201213, done.
remote: Total 201213 (delta 0), reused 0 (delta 0), pack-reused 201213
Receiving objects: 100% (201213/201213), 124.99 MiB | 2.34 MiB/s, done.
Resolving deltas: 100% (148837/148837), done.
Checking connectivity... done.
$ pushd esp-idf/
~/toit_new/esp-idf ~/toit_new
$ git checkout patch-head-4.3-3
Branch patch-head-4.3-3 set up to track remote branch patch-head-4.3-3 from origin.
Switched to a new branch 'patch-head-4.3-3'
$ git submodule update --init --recursive
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib_esp32'
Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c3_family'
Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
Submodule 'components/cbor/tinycbor' (https://github.com/intel/tinycbor.git) registered for path 'components/cbor/tinycbor'
Submodule 'components/cmock/CMock' (https://github.com/ThrowTheSwitch/CMock.git) registered for path 'components/cmock/CMock'
Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Submodule 'components/tinyusb/tinyusb' (https://github.com/espressif/tinyusb.git) registered for path 'components/tinyusb/tinyusb'
Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
Submodule 'examples/build_system/cmake/import_lib/main/lib/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
Submodule 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' (https://github.com/espressif/esp-cryptoauthlib.git) registered for path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'
Cloning into 'components/asio/asio'...
remote: Enumerating objects: 48394, done.
[... 231 lines cut ...]
Receiving objects: 100% (5869/5869), 6.62 MiB | 3.28 MiB/s, done.
Resolving deltas: 100% (3628/3628), done.
Checking connectivity... done.
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Cloning into 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'...
remote: Enumerating objects: 4460, done.
remote: Counting objects: 100% (307/307), done.
remote: Compressing objects: 100% (147/147), done.
remote: Total 4460 (delta 231), reused 192 (delta 160), pack-reused 4153
Receiving objects: 100% (4460/4460), 3.21 MiB | 3.89 MiB/s, done.
Resolving deltas: 100% (2963/2963), done.
Checking connectivity... done.
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Cloning into 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'...
remote: Enumerating objects: 695, done.
remote: Counting objects: 100% (695/695), done.
remote: Compressing objects: 100% (347/347), done.
remote: Total 695 (delta 385), reused 641 (delta 332), pack-reused 0
Receiving objects: 100% (695/695), 1.02 MiB | 0 bytes/s, done.
Resolving deltas: 100% (385/385), done.
Checking connectivity... done.
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'bb672b0437485fc7420add178299631692b15ac3'
Unable to checkout '02638065e72ea689eb0d3509b1960b31c515616e' in submodule path 'components/lwip/lwip'
$ 
```

erikcorry, 2021-12-02:
This works for me.  I ran the same commands as your dump and it worked.  You have a strange line:

```
remote: Enumerating objects: 50148, done.
remote: Counting objects: 100% (182/182), done.
remote: Compressing objects: 100% (121/121), done.
remote: Total 50148 (delta 108), reused 110 (delta 59), pack-reused 49966
Receiving objects: 100% (501Cloning into 'components/lwip/lwip'...48/50148), 9.94 MiB | 2.56 MiB/s, done.
Resolving deltas: 100% (37849/37849), done.
```

In the middle of the text "50148/50148" there is an inserted text "Cloning into 'components/lwip/lwip'..."

It's as if there are two clones going on at once, or you have a background program and a foreground program trying to do the same thing twice?

Does it work to delete everything and start over?

mikini, 2021-12-02:
You're right, that looks really strange. I did it a couple fo times, but have nothing of any sorts, that could trigger anything running in parallel. Although I pasted the 5 commands into the shell in one go, but that usually works fine.

This is on a spare machine running Ubuntu 16.04 (EOL'ed) with git version 2.7.4. On Ubuntu 20.04 it also works for me, but when I inspect the repo I can't locate the 0263 commit that the containing repo wants to checkout, consistent with the error message.

I just redid it from scratch, manually pasting the lines one-by-one and I get the same result., don't see any sign of it being accessed multiple time. My best guess ATM is that this is an "old git" artefact;

```
:~$ git clone https://github.com/toitlang/toit toit_lwip-test
Cloning into 'toit_lwip-test'...
remote: Enumerating objects: 1166, done.
remote: Counting objects: 100% (1166/1166), done.
remote: Compressing objects: 100% (915/915), done.
remote: Total 1166 (delta 352), reused 1027 (delta 226), pack-reused 0
Receiving objects: 100% (1166/1166), 4.93 MiB | 4.48 MiB/s, done.
Resolving deltas: 100% (352/352), done.
Checking connectivity... done.
:~$ cd toit_lwip-test/
:~/toit_lwip-test (master=)$ git clone https://github.com/toitware/esp-idf.git
Cloning into 'esp-idf'...
remote: Enumerating objects: 201220, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 201220 (delta 0), reused 2 (delta 0), pack-reused 201213
Receiving objects: 100% (201220/201220), 125.06 MiB | 2.83 MiB/s, done.
Resolving deltas: 100% (148821/148821), done.
Checking connectivity... done.
:~/toit_lwip-test (master %=)$ pushd esp-idf/
~/toit_lwip-test/esp-idf ~/toit_lwip-test
:~/toit_lwip-test/esp-idf (master=)$ git checkout patch-head-4.3-3
Branch patch-head-4.3-3 set up to track remote branch patch-head-4.3-3 from origin.
Switched to a new branch 'patch-head-4.3-3'
:~/toit_lwip-test/esp-idf (patch-head-4.3-3=)$ git submodule update --init --recursive
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib_esp32'
Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c3_family'
Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
Submodule 'components/cbor/tinycbor' (https://github.com/intel/tinycbor.git) registered for path 'components/cbor/tinycbor'
Submodule 'components/cmock/CMock' (https://github.com/ThrowTheSwitch/CMock.git) registered for path 'components/cmock/CMock'
Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Submodule 'components/tinyusb/tinyusb' (https://github.com/espressif/tinyusb.git) registered for path 'components/tinyusb/tinyusb'
Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
Submodule 'examples/build_system/cmake/import_lib/main/lib/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
Submodule 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' (https://github.com/espressif/esp-cryptoauthlib.git) registered for path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'
Cloning into 'components/asio/asio'...
remote: Enumerating objects: 48394, done.
remote: Counting objects: 100% (15174/15174), done.
remote: Compressing objects: 100% (1324/1324), done.
[... cut ...]
remote: Total 695 (delta 385), reused 641 (delta 332), pack-reused 0
Receiving objects: 100% (695/695), 1.02 MiB | 1.84 MiB/s, done.
Resolving deltas: 100% (385/385), done.
Checking connectivity... done.
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'bb672b0437485fc7420add178299631692b15ac3'
Unable to checkout '02638065e72ea689eb0d3509b1960b31c515616e' in submodule path 'components/lwip/lwip'
:~/toit_lwip-test/esp-idf (patch-head-4.3-3 *=)$ 
```

kasperl, 2021-12-04:
Should we close this is working as intended for now, @mikini?

mikini, 2021-12-05:
I've been looking a bit into this as I think I see an inconsistency on the Toitware fork, that is somehow mitigated by more recent git clients.

The fork is actually referencing a commit for the submodule which is part of another repository than the repository indicated by .gitmodules (Toitware vs Espressif). Whether this can lead to other problems than the one I experienced using an old git client is hard for me to predict, but to me an ambiguous description of the repositories' relationship is not good.

I'll try to substantiate this with some facts, everything done on command line is on Ubuntu 20.04/git 2.25.1.

components/lwip/lwip submodule is [cloned from espressif](https://github.com/toitware/esp-idf/blob/patch-head-4.3-3/.gitmodules#L50-L52):
```
:~/toit_new/esp-idf (patch-head-4.3-3)$ grep lwip -A2 .gitmodules 
[submodule "components/lwip/lwip"]
	path = components/lwip/lwip
	url = ../../espressif/esp-lwip.git

[submodule "components/mqtt/esp-mqtt"]
:~/toit_new/esp-idf (patch-head-4.3-3)$ 
```
However, the submodule reference [points to commit 02638065e72ea689eb0d3509b1960b31c515616e](https://github.com/toitware/esp-idf/tree/patch-head-4.3-3/components/lwip#files) which is not included on any branches of the espressif repository, but which has somehow been checked out as a detached HEAD by new git clients anyhow:
```
:~/toit_new/esp-idf (patch-head-4.3-3)$ git log -1 -p components/lwip/lwip/
commit 34edcb4ad64e21818e0d2ed482a30cb6782de9b2
Author: Erik Corry <erik@toitware.com>
Date:   Mon Nov 15 14:15:15 2021 +0100

    Roll to LWIP with unix contrib dir (#43)

diff --git a/components/lwip/lwip b/components/lwip/lwip
index c1195e00a..02638065e 160000
--- a/components/lwip/lwip
+++ b/components/lwip/lwip
@@ -1 +1 @@
-Subproject commit c1195e00a4edfa03e92eb24adb675154747f6db5
+Subproject commit 02638065e72ea689eb0d3509b1960b31c515616e
:~/toit_new/esp-idf (patch-head-4.3-3)$ 

:~/toit_new/esp-idf (patch-head-4.3-3)$ 
 git -C components/lwip/lwip/ branch --contains 02638065e72ea689eb0d3509b1960b31c515616e
* (HEAD detached at 02638065)
:~/toit_new/esp-idf (patch-head-4.3-3)$ 
```
Trying to reference this commit on a plain clone of esp-lwip will fail, but on the web interface [github will show the commit but warn that it is contained on a fork](https://github.com/espressif/esp-lwip/commit/02638065e72ea689eb0d3509b1960b31c515616e):
```
:~$ git clone git@github.com:espressif/esp-lwip
Cloning into 'esp-lwip'...
remote: Enumerating objects: 50156, done.
remote: Counting objects: 100% (190/190), done.
remote: Compressing objects: 100% (124/124), done.
remote: Total 50156 (delta 113), reused 118 (delta 64), pack-reused 49966
Receiving objects: 100% (50156/50156), 9.95 MiB | 4.28 MiB/s, done.
Resolving deltas: 100% (37854/37854), done.
:~$ cd esp-lwip/
:~/esp-lwip (2.1.2-esp)$ git show 02638065e72ea689eb0d3509b1960b31c515616e
fatal: bad object 02638065e72ea689eb0d3509b1960b31c515616e
:~/esp-lwip (2.1.2-esp)$
```
It seems that somehow the checkout of an unknown commit reference from a submodule reference will/can result in a search or traversal of forks and fetching of objects from that fork. Maybe that is why duplicated "Cloning into 'components/lwip/lwip'..." is seen when done on older git (but somehow still doesn't fetch the correct objects)?

Looking at what happens during recursive submodule update and init on git 2.25.1 actually shows a subtle extra fetch being done during checkout on components/lwip/lwip (for reference complete dump below):

```
...
Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
Submodule path 'components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
remote: Enumerating objects: 62, done.
remote: Counting objects: 100% (62/62), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 62 (delta 22), reused 60 (delta 22), pack-reused 0
Unpacking objects: 100% (62/62), 54.42 KiB | 995.00 KiB/s, done.
From https://github.com/espressif/esp-lwip
 * branch              02638065e72ea689eb0d3509b1960b31c515616e -> FETCH_HEAD
Submodule path 'components/lwip/lwip': checked out '02638065e72ea689eb0d3509b1960b31c515616e'
[... 86 lines cut ...]
remote: Enumerating objects: 62, done.
remote: Counting objects: 100% (62/62), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 62 (delta 22), reused 60 (delta 22), pack-reused 0
Unpacking objects: 100% (62/62), 54.42 KiB | 995.00 KiB/s, done.
From https://github.com/espressif/esp-lwip
 * branch              02638065e72ea689eb0d3509b1960b31c515616e -> FETCH_HEAD
Submodule path 'components/lwip/lwip': checked out '02638065e72ea689eb0d3509b1960b31c515616e'
Submodule path 'components/mbedtls/mbedtls': checked out '6465247f67167518b8813ae2faaf422704e4b1a3'
Submodule path 'components/mqtt/esp-mqtt': checked out 'f10321a53b53a146ee299cfecc320b89c0cf6611'
Submodule path 'components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
Submodule 'third-party/mruby' (https://github.com/mruby/mruby) registered for path 'components/nghttp/nghttp2/third-party/mruby'
Submodule 'third-party/neverbleed' (https://github.com/tatsuhiro-t/neverbleed.git) registered for path 'components/nghttp/nghttp2/third-party/neverbleed'
Cloning into '/home/miki/esp-idf/components/nghttp/nghttp2/third-party/mruby'...
Cloning into '/home/miki/esp-idf/components/nghttp/nghttp2/third-party/neverbleed'...
[... cut ...]
Submodule path 'components/tinyusb/tinyusb': checked out '334e95fac52a607150157ae5199a19e11f843982'
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'bb672b0437485fc7420add178299631692b15ac3'
:~/esp-idf (patch-head-4.3-3)$ 
```

kasperl, 2022-08-20:
We believe this has been fixed as part of upgrading to ESP-IDF v4.3.

closed 2022-08-20 by kasperl