toitlang/toit #88: Submodule path 'esp-idf/components/coap/libcoap/ext/tinydtls' do not contain 7f8c86e501e690301630029fa9bae22424adf618

kasperl, 2021-12-07:
Following the new instructions for getting Toitware's ESP-IDF fork:

``` sh
git submodule update --init --recursive
```

on macOS leads to this output for me:

```
Submodule path 'esp-idf': checked out '55d7da2dbe01c9a537c29e7799fb6241da9ea112'
Submodule path 'esp-idf/components/asio/asio': checked out 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
Submodule path 'esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
Submodule path 'esp-idf/components/bt/controller/lib_esp32': checked out 'fb49791b7c1a8a35f06e68124c90022667b4cff1'
Submodule path 'esp-idf/components/bt/controller/lib_esp32c3_family': checked out '12f00c45ce9c8cf9a9b2e607b4954f12d4191ffb'
Submodule path 'esp-idf/components/bt/host/nimble/nimble': checked out 'aef55bbf636ed580d4d6408a5c2e75d1f70a875e'
Submodule path 'esp-idf/components/cbor/tinycbor': checked out '085ca40781f7c39febe6d14fb7e5cba342e1804b'
Submodule path 'esp-idf/components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
Submodule path 'esp-idf/components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
Submodule path 'esp-idf/components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
Submodule path 'esp-idf/components/coap/libcoap': checked out '98954eb30a2e728e172a6cd29430ae5bc999b585'
fatal: remote error: want 7f8c86e501e690301630029fa9bae22424adf618 not valid
Fetched in submodule path 'esp-idf/components/coap/libcoap/ext/tinydtls', but it did not contain 7f8c86e501e690301630029fa9bae22424adf618. Direct fetching of that commit failed.
Submodule path 'esp-idf/components/esp_wifi/lib': checked out 'e6945e61f7c63545a77b0575c3770a85b4de948e'
Submodule path 'esp-idf/components/esptool_py/esptool': checked out '4698b396730b23fb4aab023c5fb1744db957fc4c'
Submodule path 'esp-idf/components/expat/expat': checked out '968b8cc46dbee47b83318d5f31a8e7907199614b'
Submodule path 'esp-idf/components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
Submodule path 'esp-idf/components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
Submodule path 'esp-idf/components/lwip/lwip': checked out '02638065e72ea689eb0d3509b1960b31c515616e'
Submodule path 'esp-idf/components/mbedtls/mbedtls': checked out '6465247f67167518b8813ae2faaf422704e4b1a3'
Submodule path 'esp-idf/components/mqtt/esp-mqtt': checked out 'f10321a53b53a146ee299cfecc320b89c0cf6611'
Submodule path 'esp-idf/components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
Submodule path 'esp-idf/components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
Submodule path 'esp-idf/components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
Submodule path 'esp-idf/components/protobuf-c/protobuf-c': checked out 'dac1a65feac4ad72f612aab99f487056fbcf5c1a'
Submodule path 'esp-idf/components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
Submodule path 'esp-idf/components/tinyusb/tinyusb': checked out '334e95fac52a607150157ae5199a19e11f843982'
Submodule path 'esp-idf/components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'esp-idf/examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'esp-idf/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'bb672b0437485fc7420add178299631692b15ac3'
Failed to recurse into submodule path 'esp-idf/components/coap/libcoap'
Failed to recurse into submodule path 'esp-idf'
```
```

kasperl, 2021-12-07:
Could be related to #45, @mikini?

kasperl, 2021-12-27:
Looks like this is broken on the ESP-IDF side. An upgrade to 4.4 should fix this. Until then, we can use:

```
cd $IDF_PATH/components/coap/libcoap/ext/tinydtl
git remote set-url origin https://github.com/eclipse/tinydtls.git
```

to correct the remote URL of the tinydtls component.

floitsch, 2022-07-08:
I tried with both our 4.3 and 4.4 branch and could not reproduce:
4.3: https://github.com/toitlang/toit/runs/7249126224?check_suite_focus=true
4.4: https://github.com/toitlang/toit/runs/7249062886?check_suite_focus=true

Closing for now. Feel free to reopen.