toitlang/toit #130: Clang 13 on macOS mis-compiles third_party/esp-idf/components/mbedtls

kasperl, 2021-12-13:
The Toit virtual machine compiled with Clang 13 on macOS, does not have a working TLS implementation. Clang 12.0.5 generates the correct code.

Could be related to https://github.com/ARMmbed/mbedtls/issues/5052.

kasperl, 2021-12-13:
At least one problem is here: https://github.com/espressif/mbedtls/blob/6465247f67167518b8813ae2faaf422704e4b1a3/include/mbedtls/bn_mul.h#L258.

This should be fixed like here: https://github.com/ARMmbed/mbedtls/commit/d337fbc4cb1ded1d58a1c4fe92aa9988d68ec1f3.

kasperl, 2021-12-14:
The GitHub actions we use to build releases now use Clang 12, which makes this less critical.

See 25dc42a790cabf69519b9a0c0f2a9e8581ff8795.

floitsch, 2022-07-21:
Has been fixed by upgrading ESP-IDF.

closed 2022-07-21 by floitsch