toitlang/toit #200: Driver for the DHT11/DHT22

floitsch, 2021-12-24:
It would be very useful to be able to use the DHT11/DHT22 sensors from Toit code.

Inspiration: https://github.com/adafruit/DHT-sensor-library.

Transferred from https://github.com/toitware/public/issues/16

floitsch, 2021-12-24:
Copying over @xal88's comment: 

Would also vote for the One-wire driver.

Personally I do NOT recommend the use of the DHT/AM22x sensor family. Gets broken in wet or dusty environment.
For a few cent more you can get the HTU21/Si7021 sensor with PTFE membrane cover.

A good comparison can be found here: https://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

floitsch, 2021-12-24:
Note that there is a DHT11 package:
https://pkg.toit.io/package/github.com%2Fnilwes%2FDHT11

It uses bit-banging to communicate with the chip, so it may be unreliable. (That said: it seems to work quite well).

Both DHT11 and DHT22 could benefit from some hardware support to make things easier. Therefore related to https://github.com/toitlang/toit/issues/201

krzychb, 2021-12-27:
Unfortunately, I could not make the DHT11 package work with DHT11 and AM2302 sensors I have. 
Then I realized that the BME280 sensor provides humidity measurements as well and is well supported by Toit.

Arkoxs, 2022-03-12:
I think that having a broad spectrum of supported devices will benefit the toit/jaguar platform. DHT11/DHT22 is used in quite a lot of starting projects and a lot of people have them laying around.

floitsch, 2022-03-12:
@lask is making good progress on the 1-wire protocol ( https://github.com/toitlang/toit/issues/201) using the RMT peripheral.
It should be straightforward to use the peripheral for dht11 and dht22.

closed 2022-06-30 by floitsch