toitlang/toit #201: Add 1-Wire protocol support

floitsch, 2021-12-24:
I'd like to be able to use devices like DS18B20 that communicate via the 1-Wire protocol.

It would be ideal if Toit would come with built-in support for the 1-Wire protocol, so it would be straightforward to attach and control a range of peripherals.

Transferred from https://github.com/toitware/public/issues/14

artalar, 2022-01-03:
Hi, what is the status and priority of this issue?
I can't use dht11, because of it shape. [This mode of DS18B20](https://aliexpress.ru/item/4000895660165.html?spm=a2g2w.productlist.0.0.1fe7732eTkoOHb) is much useful, and have a less price also.

floitsch, 2022-01-03:
So far nobody has started working on it, but it's very high on our priority list.

sebastianneubauer, 2022-02-23:
Good to hear that this is on the roadmap! Naive and a complete noob that I am in microcontrollers and even more in Toit, I thought: I will just implement 1-wire myself before luckily finding this issue here. Inspired by the very simple and straightforward implementation in MicroPython (https://github.com/micropython/micropython/blob/5f50f4a130e60c4bc5f4fbf58ac849dcaf764e72/drivers/onewire/onewire.py), I was thinking: how hard can it be? Sadly I could not get it working, reminding me on Dunnning-Kruger. I am pretty sure it is a whole bunch of layer-7 errors and I guess I would be better off just waiting for the expert implementation, but I like to learn.

First thing I have no clue what the correct configuration of the gpio is. The official implementation (https://www.maximintegrated.com/en/design/technical-documents/app-notes/1/126.html e.g. the reset command) says:
* Delay G
* Drive bus low, delay H
* Release bus, delay I
* Sample bus, 0 = device(s) present, 1 = no device present
* Delay J
 
My dump first implementation was this:
```
import gpio

A := 6
B := 64
C := 60
D := 10
E := 9
F := 55
G := 0
H := 480
I := 70
J := 410

class OWsensor:
                           
  data_pin_       := ?

  // We need one pin for reading and one pin for writing since using the same pin and switching it
  // between being input and output takes too long.
  constructor data_pin/int:
    data_pin_     = gpio.Pin data_pin --output --pull_up

  reset:
    result :=0
    
    beginning := Time.monotonic_us
    sleep (Duration --us=G)
    data_pin_.config --output
    data_pin_.set 0
    
    time_g := Time.monotonic_us
    sleep (Duration --us=H)
    data_pin_.config --output  --open_drain
    data_pin_.set 1
    time_h := Time.monotonic_us
    sleep (Duration --us=I)
    data_pin_.config --input
    result = data_pin_.get
    time_I := Time.monotonic_us
    sleep (Duration --us=J)
    time_all := time_I - beginning
    print("g $(time_g-beginning) h $(time_h-time_g) i $(time_I-time_h) overall $time_all")
    return result
```
So, as I said, I am sure I got the "open drain" gpio thing completely wrong, how would that be correctly implemented?
What baffles me in this implementation, is the very off timing (note, this is microseconds, µs):
```
[example.toit] g 23 h 1371 i 468 overall 1862
[example.toit] g 23 h 1371 i 457 overall 1851
[example.toit] g 24 h 1364 i 468 overall 1856
[example.toit] g 23 h 3314 i 661 overall 3998
[example.toit] g 23 h 1365 i 468 overall 1856
```
As you can see, the times are way off. Time H should be 480µs but it is over 1300µs and even worse: with a very huge spread!
Am I doing something wring, or can this fine timing not be achieved using the Toit language? (e.g. does it have to be implemented in native code and wrapped by Toit?)
Thanks for any hints, and thanks for working on this awesome project!

floitsch, 2022-02-24:
You correctly identified the difficulty: these fine timings can't be done in Toit.
There are two ways to work around it:
- have a c/c++ primitive that does the work for you with tight timing controls
- use an EPS32 peripheral for it.

@lask has started work on the latter. By using the infrared peripheral we should be able to achieve precise timings without hogging the CPU.

artalar, 2022-02-24:
Can't wait for it too 🤞

closed 2022-06-30 by floitsch