toitlang/toit #218: ESP32-CAM

floitsch, 2021-12-29:
Support the ESP32-CAM.

- [ ] Taking pictures and constantly uploading them
- [ ] Taking pictures and saving them locally in the microSD card, to be queried and downloaded remotely
- [ ] Videos
- [ ] Image processing (C bridge?)

dumblob, 2023-06-27:
On a similar note, I am looking for support for **two** non-MEMS microphones (yep, I need the traditional analogue signal to be "ADCed") being mixed into one stereo-signal which I can then both in real-time transfer over network as well as save the very same stream to a flash memory.

Any plans to support such use-cases?

floitsch, 2023-06-27:
@mikkeldamsgaard do you know what is already possible and what would need to happen for audio signals?

mikkeldamsgaard, 2023-06-30:
@floitsch @dumblob The ESP32 has two ADC's that are supported by toit already and the flash is also supported. The issue will be to get a constant sample rate, as the ADC's are not currently supported in toit on the I2S.

The specification for the ADC's in the esp32 are not optimal for audio sampling anyway (12-bit and very bad SNR), so if you want good quality digital recordings, I would recommend adding a dedicated audio codec/adc to the system and interface that to the ESP32 via I2S.

dumblob, 2023-06-30:
> The specification for the ADC's in the esp32 are not optimal for audio sampling anyway (12-bit and very bad SNR)

Oh, thanks for the nudge! I would not have thought of this myself.