toitlang/toit #252: Error message through LSP is poorly formatted

kasperl, 2022-01-03:
In VS Code, I get errors like this:

```
Argument mismatch for 'bar'Too few arguments provided
```

Not sure where the problem stems from. Maybe it is a newline handling thing?

lask, 2022-01-04:
[The newline is definitely added in the compiler](https://github.com/toitlang/toit/blob/9acf6e2a949df699b1a2366523dbd526df6922e7/src/compiler/no_such_method.cc#L170). VS Code does ignore newlines in a number of places, so that is probably the problem.

floitsch, 2022-01-04:
Yes.
Erik asked me to add the newlines to make the error messages look better on the console.
I forgot to check how they were looking in vscode.
The easiest is probably to just replace newlines with ". " (dot, spaces) When sending them to the LSP client.
Alternatively I could also just revert the newline error message changes.

lask, 2022-01-04:
It looks like VS Code doesn't ignore newlines. I tested it with the VS Code sample LSP server found here https://github.com/microsoft/vscode-extension-samples/tree/main/lsp-sample by adding newlines to the error messages. This worked fine, so it would seem that it is possible to display error messages with newlines in VS Code.

floitsch, 2022-01-04:
Could easily be that the LSP protocol requires \r\n.

closed 2022-01-05 by lask