toitlang/toit #302: Implement ESP-now

floitsch, 2022-01-13:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html

nivoc, 2022-08-08:
I'm willing to support that if we find someone to do it.

floitsch, 2022-08-08:
It's hard to estimate.
There are only very few esp-now functions and they aren't too complex. I'm pretty sure that it's pretty simple to get those functions exposed to Toit.
The harder part is to deal with the integration of the WiFi. For example, we have abstracted away networking so that drivers (WiFi and others) run in their own services. However, esp-now needs to have WiFi enabled to work. This is probably something that could be ignored in the first version, but needs to be tackled eventually.

The other (minor) difficulty is that the esp-now functions don't use queues yet. This means that we would need to create them (using the existing queue-mechanisms) ourselves. For sending we could probably just ignore whether the sending was successful or not (fire-and-forget), but for receiving we would definitely need these queues.

All in all, it looks definitely feasible. I will see if it can rise to the top of my TODO list.

nivoc, 2022-08-08:
Cool Thanks for the Details an Todo-Push :-)! Yes a rough implantation would be good enough in the first version. Could be improved if people really start to use it.

floitsch, 2023-01-02:
Espnow was implemented in https://github.com/toitlang/toit/commit/7722235e7786d63ab8e5d496527c43eb13ee0eaa

closed 2023-01-02 by floitsch