toitlang/toit #434: Cannot receive data from stream

snxx-lppxx, 2022-02-09:
```sh
$ toit execute main.toit
8.93 MiB / 8.93 MiB [----------------------------------] 100.00% 328.10 KiB p/s

Successfully installed Toit SDK (version: v1.6.0) locally.
Cannot receive data from stream: rpc error: code = Unknown desc = lstat /tmp/sta
ging/565a1fbf-8e2e-4fc8-b4d8-d5c7d6fc2e6e/sdk/toitc: no such file or directory
rpc error: code = Unknown desc = lstat /tmp/staging/565a1fbf-8e2e-4fc8-b4d8-d5c7
d6fc2e6e/sdk/toitc: no such file or directory
```

Operating system: Windows 7.

floitsch, 2022-02-09:
Thanks for the report.
While we are working on fixing this, try the downloadable binaries from toitlang/toit could be a work-around. Especially on Windows they aren't as feature-complete yet, but maybe they get you going.

DanNduati, 2022-02-13:
I have the same issue on Ubuntu 18.04.6 LTS
```bash
$ toit run --no-device hello.toit 
Login required, start SSO login
8.60 MiB / 8.60 MiB [------------------------------------] 100.00% 66.48 KiB p/s

Successfully installed Toit SDK (version: v1.6.0) locally.
Cannot receive data from stream: rpc error: code = Unknown desc = lstat /tmp/staging/7f8fc092-802a-4065-9840-f1cbd44d0832/sdk/toitc: no such file or directory
rpc error: code = Unknown desc = lstat /tmp/staging/7f8fc092-802a-4065-9840-f1cbd44d0832/sdk/toitc: no such file or directory
```
```bash
toit version
+---------+------------+
| VERSION |    DATE    |
+---------+------------+
| v1.18.2 | 2022-02-05 |
+---------+------------+
```

closed 2022-02-15 by erikcorry

floitsch, 2022-02-15:
The fix has been merged and will be visible as soon as we make a release.

DanNduati, 2022-02-15:
Great!