toitlang/toit #473: What are Bytes allocated in object heap ?

muhlpachr, 2022-02-19:
According https://libs.toit.io/core/utils/library-summary#process_stats process_stats shell return Bytes allocated in object heap.
It seams, my program allocate more bytes than esp32 has RAM available.
Why Bytes allocated in object heap increases continuously ?
```
jag version
Version:         v1.0.7
SDK version:     v1.6.7
Build date:      2022-02-18T14:26:00Z
```
```
main:
  i := 0
  while true:
    if i % 100 == 0:
      print "Test: $i GC#: $gc_count Heap: $process_stats[4]"
    i++
    sleep --ms=1
```
```
Test: 0 GC#: 0 Heap: 1788
Test: 100 GC#: 0 Heap: 4832
Test: 200 GC#: 0 Heap: 4980
...
Test: 9500 GC#: 0 Heap: 19424
Test: 9600 GC#: 0 Heap: 19580
Test: 9700 GC#: 0 Heap: 19736
Test: 9800 GC#: 0 Heap: 19892
Test: 9900 GC#: 0 Heap: 20048
Test: 10000 GC#: 1 Heap: 20204
Test: 10100 GC#: 1 Heap: 20360
Test: 10200 GC#: 1 Heap: 20516
Test: 10300 GC#: 1 Heap: 20672
Test: 10400 GC#: 1 Heap: 20828
Test: 10500 GC#: 1 Heap: 20984
...
Test: 21400 GC#: 1 Heap: 37988
Test: 21500 GC#: 1 Heap: 38144
Test: 21600 GC#: 1 Heap: 38300
Test: 21700 GC#: 2 Heap: 38456
Test: 21800 GC#: 2 Heap: 38612
Test: 21900 GC#: 2 Heap: 38768
...
Test: 684800 GC#: 59 Heap: 1096284
Test: 684900 GC#: 59 Heap: 1096444
Test: 685000 GC#: 60 Heap: 1096604
Test: 685100 GC#: 60 Heap: 1096764
...
```

muhlpachr, 2022-02-19:
Thank you for explanation and Q&A reference, I was not aware of that.

kasperl, 2022-02-20:
Moved to Q&A: https://github.com/toitlang/toit/discussions/476.

closed 2022-02-20 by kasperl