toitlang/toit #474: App crashes when trying to resolve unknown host

rbedemann, 2022-02-19:
Hey,

I faced this issue while trying to connect my mqtt client to a broker. Due to a wrong config, I tried to connect to an unknown host which resulted in an unexpected error. I was able to identify the root cause, which seems to be located in the `net` package.

### Steps to reproduce:
Install an app on a device (tested with firmware v1.6.6) with following code:
```
import log
import net

main:
    // works:
    log.info (net.open.resolve "toit.io").stringify

    // doesn't work and crashes app:
    log.info (net.open.resolve "thishostisnotknowntoanyone").stringify
```

### Expected behaviour:
`net.open.resolve` returns the resolved IP for known hosts and resturns an empty list of ip addresses for unknown hosts.

### Actual behaviour:

First call to `net.open.resolve` returns an IP address as expected. Second call to resolve an unknown host results in following error:
```
EXCEPTION error. 
WRONG_OBJECT_TYPE
  0: process_send_             <sdk>/core/message.toit:42:3
  1: RpcRequest_.process       <sdk>/rpc/broker.toit:104:7
  2: RpcRequestQueue_.ensure_processing_task_.<lambda>.<block>.<block> <sdk>/rpc/broker.toit:207:20
  3: RpcRequestQueue_.ensure_processing_task_.<lambda>.<block> <sdk>/rpc/broker.toit:201:9
  4: RpcRequestQueue_.ensure_processing_task_.<lambda> <sdk>/rpc/broker.toit:196:56
```

kasperl, 2022-02-20:
The problem was introduced in https://github.com/toitlang/toit/commit/73dc0257ae64c355bcaad21c02be8d23fe660924. The RPC mechanism doesn't know how to serialize the `DnsException` instances.

closed 2022-02-22 by kasperl

kasperl, 2022-02-23:
With v1.6.8 the exception will be:

```
EXCEPTION error. 
DNS lookup exception Server responded: NO_SUCH_DOMAIN
```

It isn't the prettiest exception - but it is now the expected message.