toitlang/toit #475: App crashes when subscribing to a device:* pubsub topic

rbedemann, 2022-02-19:
Hey,

I was playing around with the local pubsub mechanism as it is described in the [tutorial](https://docs.toit.io/tutorials/pubsub). However, when I deploy my subscriber app, I'm receiving following exception after ~7 seconds:

```
EXCEPTION error. 
COAP ERROR 132: failed to find subscription with ID: 00000000-0000-0000-0000-000000000000
  0: Client.read_response_     <sdk>/coap/client.toit:177:7
  1: Client.stream_read        <sdk>/coap/client.toit:82:12
  2: Stream_.receive           system/kernel/console/v2_client.toit:201:20
  3: PubSubFetch_.receive      system/kernel/console/v2_client.toit:137:20
  4: ConsoleConnector.fill_pubsub_queues_.<block>.<block> system/kernel/console_connector.toit:542:58
  5: ConsoleConnector.fill_pubsub_queues_.<block> system/kernel/console_connector.toit:539:5
  6: ConsoleConnector.fill_pubsub_queues_ system/kernel/console_connector.toit:520:3
  7: ConsoleConnector.process_pubsub_.<block> system/kernel/console_connector.toit:513:7
  8: ConsoleConnector.process_pubsub_ system/kernel/console_connector.toit:510:3
  9: ConsoleConnector.run.<block>.<block>.<block>.<block> system/kernel/console_connector.toit:275:17
 10: Task_.with_deadline_.<block> <sdk>/core/task.toit:79:21
 11: Task_.with_deadline_      <sdk>/core/task.toit:73:3
 12: with_timeout              <sdk>/core/utils.toit:174:15
...
 17: catch                     <sdk>/core/exceptions.toit:87:10
 18: ConsoleConnector.run.<block>.<block> system/kernel/console_connector.toit:257:16
 19: ConsoleConnector.run.<block> system/kernel/console_connector.toit:235:37
 20: Connection.with_client.<block>.<block>.<block> system/kernel/connection.toit:48:16
 21: Connection.with_client.<block>.<block> system/kernel/connection.toit:45:7
 22: Connection.with_client.<block> system/kernel/connection.toit:44:15
 23: ASyncComponent.do.<block> system/kernel/component.toit:131:19
 24: ASyncComponent.do         system/kernel/component.toit:128:3
 25: Connection.with_client    system/kernel/connection.toit:44:13
 26: ConsoleConnector.run      system/kernel/console_connector.toit:235:10
 27: ConsoleSyncJob.run_.<block> system/kernel/scheduler.toit:419:26
 28: ConsoleSyncJob.run_       system/kernel/scheduler.toit:415:3
 29: ConsoleSyncJob.start.<lambda>.<block>.<block> system/kernel/scheduler.toit:410:11
 30: catch.<block>             <sdk>/core/exceptions.toit:114:10
 31: catch                     <sdk>/core/exceptions.toit:112:1
 32: catch                     <sdk>/core/exceptions.toit:87:10
 33: ConsoleSyncJob.start.<lambda>.<block> system/kernel/scheduler.toit:409:9
 34: ConsoleSyncJob.start.<lambda> system/kernel/scheduler.toit:407:25
```

Unfortunately this makes the entire app crash. It neither works if I execute the code via the toit console nor when I deploy my code as a job. I also added the subscription to my *.yaml file as it otherwise complained that the PUBSUB topic was not set up. Also with `device-memory:` prefix it did not work.

I'm clueless and have no idea what I could have done wrong...

kasperl, 2022-02-20:
Hi @rbedemann! Can you share the code that causes this problem to appear including the `.yaml` files? It looks like we end up asking the cloud for an non-existing pubsub subscription.

rbedemann, 2022-02-20:
Hey @kasperl,

it's actually the exact same code as from the tutorial which I took (see example1 below). However, without the `device` pubsub subscription listed in the `.yaml` file, I receive the error of a missing subscription setup. This is how I ended up adding this subcription to the list in the `.yaml` file which led me to the above mentioned problem (see example2 below).

-----
## Example 1
`device:` subscription not listed in `.yaml`. Same code from tutorial:

### example1.toit
```
import pubsub
topic ::= "device:hello-world"

main:
  pubsub.subscribe topic: | msg/pubsub.Message |
    print "Received message '$msg.payload.to_string'"
    // Stop listening for other messages and exit 'main'.
    return
```
### example1.yaml 
```
name: example1
entrypoint: example1.toit
```

### Error:
```
EXCEPTION error. 
PUBSUB_SUBSCRIPTION_NOT_SETUP
  0: PubSubManager.read_bytes  system/kernel/pubsub.toit:228:7
  1: PubSubApi.bind.<lambda>   system/api/pubsub.toit:31:23
  2: RpcRequest_.process.<block> <sdk>/rpc/broker.toit:98:26
  3: RpcRequest_.process       <sdk>/rpc/broker.toit:95:3
  4: RpcRequestQueue_.ensure_processing_task_.<lambda>.<block>.<block> <sdk>/rpc/broker.toit:207:20
  5: RpcRequestQueue_.ensure_processing_task_.<lambda>.<block> <sdk>/rpc/broker.toit:201:9
  6: RpcRequestQueue_.ensure_processing_task_.<lambda> <sdk>/rpc/broker.toit:196:56
```
---
## Example 2
`device:` subscription listed in `.yaml` (to avoid error seen in example1). Same code from tutorial:

### example2.toit
```
import pubsub
topic ::= "device:hello-world"

main:
  pubsub.subscribe topic: | msg/pubsub.Message |
    print "Received message '$msg.payload.to_string'"
    // Stop listening for other messages and exit 'main'.
    return
```
### example2.yaml 
```
name: example2
entrypoint: example2.toit
pubsub:
  subscriptions:
    - "device:hello-world"
```
### Error:
```
EXCEPTION error. 
COAP ERROR 132: failed to find subscription with ID: 00000000-0000-0000-0000-000000000000
  0: Client.read_response_     <sdk>/coap/client.toit:177:7
  1: Client.stream_read        <sdk>/coap/client.toit:82:12
  2: Stream_.receive           system/kernel/console/v2_client.toit:201:20
  3: PubSubFetch_.receive      system/kernel/console/v2_client.toit:137:20
  4: ConsoleConnector.fill_pubsub_queues_.<block>.<block> system/kernel/console_connector.toit:542:58
  5: ConsoleConnector.fill_pubsub_queues_.<block> system/kernel/console_connector.toit:539:5
  6: ConsoleConnector.fill_pubsub_queues_ system/kernel/console_connector.toit:520:3
  7: ConsoleConnector.process_pubsub_.<block> system/kernel/console_connector.toit:513:7
  8: ConsoleConnector.process_pubsub_ system/kernel/console_connector.toit:510:3
  9: ConsoleConnector.run.<block>.<block>.<block>.<block> system/kernel/console_connector.toit:275:17
 10: Task_.with_deadline_.<block> <sdk>/core/task.toit:79:21
 11: Task_.with_deadline_      <sdk>/core/task.toit:73:3
 12: with_timeout              <sdk>/core/utils.toit:174:15
...
 17: catch                     <sdk>/core/exceptions.toit:87:10
 18: ConsoleConnector.run.<block>.<block> system/kernel/console_connector.toit:257:16
 19: ConsoleConnector.run.<block> system/kernel/console_connector.toit:235:37
 20: Connection.with_client.<block>.<block>.<block> system/kernel/connection.toit:48:16
 21: Connection.with_client.<block>.<block> system/kernel/connection.toit:45:7
 22: Connection.with_client.<block> system/kernel/connection.toit:44:15
 23: ASyncComponent.do.<block> system/kernel/component.toit:131:19
 24: ASyncComponent.do         system/kernel/component.toit:128:3
 25: Connection.with_client    system/kernel/connection.toit:44:13
 26: ConsoleConnector.run      system/kernel/console_connector.toit:235:10
 27: ConsoleSyncJob.run_.<block> system/kernel/scheduler.toit:419:26
 28: ConsoleSyncJob.run_       system/kernel/scheduler.toit:415:3
 29: ConsoleSyncJob.start.<lambda>.<block>.<block> system/kernel/scheduler.toit:410:11
 30: catch.<block>             <sdk>/core/exceptions.toit:114:10
 31: catch                     <sdk>/core/exceptions.toit:112:1
 32: catch                     <sdk>/core/exceptions.toit:87:10
 33: ConsoleSyncJob.start.<lambda>.<block> system/kernel/scheduler.toit:409:9
 34: ConsoleSyncJob.start.<lambda> system/kernel/scheduler.toit:407:25
```

Hope that helps!

kasperl, 2022-02-22:
I certainly does help. I can reproduce the problem and have tracked it down to an issue introduced late last year. Thanks! I hope to have a fix out within a week or so.

kasperl, 2022-02-23:
Will be fixed in v1.6.8.

kasperl, 2022-02-23:
I expect v1.6.8 to be out early next week after some stress testing!

kasperl, 2022-02-23:
@rbedemann: Reach out to me via Discord and I can give you access to the alpha release right away!

https://discord.gg/Q7Y9VQ5nh2

kasperl, 2022-03-02:
I believe this has been fixed in v1.6.9 which was just released.

closed 2022-03-02 by kasperl