toitlang/toit #496: Disable Default Wifi-Connect

nivoc, 2022-02-22:
Is there a Way to not boot with Wifi/Cloud?

I have a sensor (currently micropython) that sleeps most of the time. It wakes every minute and checks the temperature and sleeps again. It should only connect to the internet if the temperature is out a defined bound. 

To connect every Minute is unnecessary and drains the battery a lot. 

It's enough to connect if there is something wrong with the temperature or every 6 hours to tell the cloud (still working - everything normal)

floitsch, 2022-02-22:
I'm assuming you are using the open-source version with Jaguar. The toit.io version has this feature integrated (with a `max_offline` configuration exactly as you describe).

Currently Jaguar connects immediately to the WiFi which is the reason the WiFi is turned on.
We have lots of plans with Jaguar, so we should be able to cover your use-case in the future.

In the meantime you could use the Toit framework without Jaguar. If you follow the steps in https://github.com/toitlang/toit#build-for-esp32 with the ESP_ENTRY set to your program. That would come with the cost of not having Jaguar (and its easy program installation) on it anymore...

nivoc, 2022-02-22:
Oh - Sorry!

That was my fault. I was looking at the wrong place. Now I see it. Its part of the deviceconfig. https://docs.toit.io/platform/devices/deviceconfig

I'm using/testing with the Cloud Version. The feature is better than I hoped for. I was looking for a function/way to control it within my software. But configuration is even better 👍 

But thanks for the detailed Jaguar Description.

I just discovered toit. I like it! Especially the "Deploy without Bricking"-Part :-)

closed 2022-02-22 by nivoc