toitlang/toit #577: Set: basic operations inconsistencies

lutzbickhardt, 2022-03-17:
Hi,

consider the following program which creates Set variables. 

a) some operations are obviously allowed for a single item, but not for multiple items; so why allow it for a single item at all?
b) some do not make sense at all (to me): such as true/false or null in a Set
c) basically what I do derive from the test is: only <strings> and <numbers>can be used in a Set.
d) so either you should update the doc (and the compiler) and/or extend the operations allowed on a Set

You should test also with .add and .remove operations.
You should test also with List and Map.
(I think when looking at the code you get the idea what to do)

Uncommenting raises the exception, all uncommented lines do not cause an exception.

main:
    exception := catch:
        sett/Set := {}

        sett = {true}
        //sett = {true, false}  => EXCEPTION -> LOOKUP_FAILED ...
        sett = {1}
        sett = {1, 2}

        sett = {"a"}
        sett = {"a", "b"}
        sett = {"a", 1}

        //sett = {"a", true}    => EXCEPTION -> LOOKUP_FAILED ...
        //sett = {1, true}      => EXCEPTION -> LOOKUP_FAILED ...

        sett = {null}
        //sett = {null, ""}     => EXCEPTION -> LOOKUP_FAILED ...
        //sett = {null, "a"}    => EXCEPTION -> LOOKUP_FAILED ...
        //sett = {null, 1}      => EXCEPTION -> LOOKUP_FAILED ...

        sett = {[2, 2, 2]}
        sett = {[2, 2, 2], [2, 2, 2]}
        //sett = {[2, 2, 2], [1, 1, 1]} => EXCEPTION -> LOOKUP_FAILED ...
    
        sett = {{2}}
        sett = {{2, 2}}
        sett = {{2}, {2}}
        sett = {{2, 2}, {2, 2}}
        //sett = {{1}, {2}}
        //sett = {{1, 1}, {2, 2}}

        sett = {[2]}
        sett = {[2, 2]}
        sett = {[2], [2]}
        sett = {[2, 2], [2, 2]}
        //sett = {[1], [2]}      => EXCEPTION -> LOOKUP_FAILED ...
        //sett = {[1, 1], [2, 2]} => EXCEPTION -> LOOKUP_FAILED ...
        
        sett = {{"x": "X"}}
        //sett = {{"x": "X"}, {"y": "Y"}} {"x": "X"}   => EXCEPTION -> LOOKUP_FAILED ...
        //sett = {{"x": "X"}, 1} => EXCEPTION -> LOOKUP_FAILED ...

    if exception:
        print "EXCEPTION -> $exception ..."

    else:
        print "D O N E !!!"

    print "test 1"

lutzbickhardt, 2022-03-17:
sett = {true} also works without exception

lutzbickhardt, 2022-03-17:
//sett = {{"x": "X"}, {"y": "Y"}} {"x": "X"}   => EXCEPTION -> LOOKUP_FAILED ...
should read
    //sett = {{"x": "X"}, {"y": "Y"}}  => EXCEPTION -> LOOKUP_FAILED ...

lutzbickhardt, 2022-03-18:
c) should read:
   c) basically what I do derive from the test is: only string and number types can be used in a Set.

erikcorry, 2022-03-18:
The Set and Map implementations use hashing to do their operations in constant speed. As a consequence their keys must have a hash_code member.

For very small sets and maps the hash_code method is sometimes not called, so the sets and maps may appear to work until more members are added.

The documentation should certainly make this more clear.

Read more at https://docs.toit.io/language/listsetmap

lutzbickhardt, 2022-03-18:
that is understood by the code LOOKUP_FAILED, but even if the hashing was called for a single member, it should not be up to the runtime to raise an exception. This can and should be detected at compile time.

floitsch, 2022-03-18:
There are a few different points here:
1. maps and sets do accept objects without `hash_code` for small sets/maps.
2. there is no type check to ensure that objects have the correct type
3. the compiler doesn't give any feedback on whether an object will work
4. the compiler doesn't give any warning for set/map literals.

I suggest the following action items:
- a. call `hash_code` even if the set/map is small. Just to make sure that a user doesn't get the impression that the container works for the given data-type. We don't need to use it, but at least there would be an error early, instead of when the map/set grows.
- b. add a `Hashable` interface (or similar)
- c. let the compiler check that objects in literals have types with `hash_code` methods. This is automatically happening once (b) is done.

This won't catch every possible case, as Toit is dynamically typed, but would probably catch many.
However, adding a `Hashable` interface and requiring it for sets and maps would be a breaking change, so can't be enforced immediately. However, (a) and (c) should be doable without breaking (most) programs.

erikcorry, 2022-03-18:
https://github.com/toitlang/toit/pull/578/files here is at least an improvement to https://libs.toit.io/core/class-Set

floitsch, 2022-03-18:
I forgot: another approach is to have an identity-hashcode by default. This way every object would be allowed.

lutzbickhardt, 2022-03-18:
I fell victim to your case a. (" ... user doesn't get the impression that the container works for the given data-type")
So whatever helps to avoid this is appreciated.

In the case of type Boolean, Set does not make sense at all (as with null).

floitsch, 2022-03-18:
There are cases where `true`, `false` and `null` would be useful:
Say you have an API that returns `null` for when the module isn't available, and integer error codes otherwise.
You might want to avoid printing error messages multiple times and keep a set for all errors you already printed. In that case, having `null` would make sense.

Similarly, there are cases where `true` or `false` would make sense. However, only if it would be mixed with other types.

Anyway, they currently don't have a hash_code and therefore won't work.

@erikcorry just wrote a PR that will at least make "bad" types fail immediately. I will see how easy it is to add support for literals.

lutzbickhardt, 2022-03-18:
if true/false was allowed only 4 combinations (TT, FF, TF, FT) would be allowed, but together with other types: ok!

Otherwise: cool!

lutzbickhardt, 2022-03-20:
and of course it could only be  {T}, {F}, {T,F}, {F,T} NOT (as I implied above) {T, T} {F, F} {T,F} {F, T)

closed 2022-06-30 by floitsch