toitlang/toit #809: The SPI module for the ESP32 does not allow you to choose if the CS pin should be held after transmission

MikAtCosesy, 2022-06-22:
There are many different IC's that use SPI as a communication method, and they have different implementations of the protocol.
Some implementations will give different responses to the same command based on the previous command and the time since the last command.

An example of this is the Si4455 radio transceiver (https://www.silabs.com/documents/public/data-sheets/Si4x55-C.pdf).
In the SPI implementation for the transceiver, you have to poll the chip's status command until it is ready. When the status commands indicate that the chip is ready, data can be read from the chip.

The current implementation of the SPI driver allows you to select a CS pin that can be controlled with the SPI peripheral register. However the exposed primitives for the SPI driver does not allow you to select wether or not the CS line should be held after a transmission. Therefore you must manually control the CS pin in such applications:

    poll_cts_ -> int:

        data := #[READ_CMD_BUFF_, 0xFF]
        
        nsel_.set 0
        device_.transfer data --read=true

        return data[1]

    read_response_ data/ByteArray:
        cts := 0

        data.fill 0xFF

        while cts != 0xFF:
            cts = poll_cts_
            if cts == 0xFF:
                device_.transfer data --read=true
            nsel_.set 1

The ESP32 has a bit that controls whether or not to hold the CS line after the given transmission. 
This flag is located in the PI_PIN_REG (0x34) as described in Register 7.14 on page 136 in the ESP32 reference manual (https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf).

This should be simple to add to the SPI transfer primitive and if a default is added to the SPI implementation in the library, it should be possible to add this feature while keeping the backward compatibility for the existing code base.

floitsch, 2022-06-23:
If I understand correctly, you would like to:
``` toit
poll_cts_ -> int:
  data := #[ READ_CMD_BUFF_, 0xFF ]
  device_.transfer data --read --keep_cs_active
  
  return data[1]

...
```

Any `transfer` that doesn't have the flag would automatically set the bit to false, thus ending the CS.

Does that sound right?

cakemaster9001, 2022-06-25:
Yes, that is exactly correct. This will allow all CS-related actions to be handled in the driver instead of the user having to do it manually.

floitsch, 2022-06-30:
Turns out that this is mostly exposed in a recent ESP-IDF version.
We will update to ESP-IDF 4.4 first, and then it should be relatively straight-forward to implement. (The hard part is the upgrading).

MikAtCosesy, 2022-06-30:
Perfect! Looking forward to it

closed 2022-07-27 by floitsch

floitsch, 2022-07-27:
This change was committed and will be available in the next release.

Note that the esp-idf currently does not give a way to deactive the CS (nor does it do it automatically). This means that it's important to finish the operation with a transmit that does not keep the CS active.

See https://github.com/espressif/esp-idf/issues/9390

PR #879 (2022-07-27): Add a way to keep CS active.
Fixes #809.
commit 9b781f21 (2022-07-27): Add a way to keep CS active. (#879)

Fixes #809.