toitlang/toit #900: Manual shutdown for Sequans cellular modem on Gpy before going into deep sleep

HoferChristoph, 2022-07-28:
Is it possible to manually shutdown the Sequans cellular modem on a Gpy before going into deep sleep although Toit is still using it?

Currently I can only put the esp32 chip on the Gpy into deep sleep which leaves the modem in some funky state and does not minimize the power consumption. Furthermore, it appears that after deep sleep the Gpy’s behaviour doesn’t seem very stable. Probably due to the state of the modem after deep sleep

When using the onboard Wi-Fi-Chip however deep sleep works as expected and really minimizes power consumption.

kasperl, 2022-07-29:
Hi @HoferChristoph. We try to close the modem by asking it to go offline (through the CFUN AT command), but maybe we're missing something. I will take a look next week.

kasperl, 2022-07-29:
In most of the other modem drivers, we manually have to power-on and power-off the modems, but we do not do that with the Monarch.

kasperl, 2022-08-03:
It is possible to use AT+SQNSSHDN, but I'm not completely sure how to reactivate the modem when waking up.