toitlang/toit #974: BLE Documentation

nivoc, 2022-08-21:
Andreas Spiess complains in its Youtube-Review that the Documentation contains a Bluetooth Tutorial but misses to explain or show how to actually transmit data - he failed there (Got only manufacturer data to work)

I had the same issue. Some days ago I discovered that the repo contains a perfect and dead simple example. 

https://github.com/toitlang/toit/blob/master/examples/ble/heart_rate.toit

Easy fix: Add a one liner to the docs

"Find a working example here: https://github.com/toitlang/toit/blob/master/examples/ble/heart_rate.toit" 

at the end of the BLE Tutorial https://docs.toit.io/tutorials/ble

:-)

kasperl, 2022-08-22:
Thanks, @nivoc! Will try to roll this out today :)

kasperl, 2022-08-22:
https://docs.toit.io/tutorials/ble has been updated. Thanks again!

closed 2022-08-22 by kasperl

nivoc, 2022-08-22:
:-)

askdesigners, 2022-10-21:
I'm having trouble getting the example to compile actually. 

```
[servo_ble.yaml] Failed deploying job 'Servo BLE' on 'home-sensor'
[servo_ble.yaml] ExitError - exit code: 1
[servo_ble.yaml] /Users/ryan/Documents/Personal/servos_app/servosdevice/servo_ble.toit:13:9: error: Unresolved identifier: 'BLE_CONNECT_MODE_UNDIRECTIONAL'
[servo_ble.yaml]   print BLE_CONNECT_MODE_UNDIRECTIONAL
[servo_ble.yaml]         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[servo_ble.yaml] /Users/ryan/Documents/Personal/servos_app/servosdevice/servo_ble.toit:14:13: error: Unresolved identifier: 'ServerConfiguration'
[servo_ble.yaml]   config := ServerConfiguration
[servo_ble.yaml]             ^~~~~~~~~~~~~~~~~~~
[servo_ble.yaml] /Users/ryan/Documents/Personal/servos_app/servosdevice/servo_ble.toit:29:20: error: Argument mismatch for 'default'
```

Is this some path issue?

askdesigners, 2022-10-21:
I've actually found both BLE and Websockets have confusingly two versions floating around. It looks like the example linked here doesn't match up with the lib as described on the SDK site here https://libs.toit.io/ble/library-summary

Is this a new/different version?? It looks like it doesn't export the same members. However when cmd + click the lib name in VSCode, I see the library as I expect to from the example...

floitsch, 2022-10-21:
That's on us...
The libs.toit.io document still points to the v1 version of Toit.
I will see if I can move the documentation to v2.
Note, that we are in the process of landing a bigger change to BLE (with the nice advantage that BLE then also works from macos).

floitsch, 2022-10-21:
> I've actually found both BLE and Websockets have confusingly two versions floating around. It looks like the example linked here doesn't match up with the lib as described on the SDK site here https://libs.toit.io/ble/library-summary
> 
> Is this a new/different version?? It looks like it doesn't export the same members. However when cmd + click the lib name in VSCode, I see the library as I expect to from the example...
> 
> Digging into this more, it looks like VSCode is referencing the Jaguar lib, while the compiler is referencing the toit one?? Do I need to reinstall toit?

We generally recommend to run with Toit v2 nowadays. This is the one that comes bundled with Jaguar.
Feel free to reach out on chat if you need more guidance or have questions on the differences of v1 and v2.

askdesigners, 2022-10-21:
ah ok gotcha. I will uninstall I guess and reinstall? I've had it a while :D

askdesigners, 2022-10-21:
Is there a workflow to upgrade to 2.0?

floitsch, 2022-10-21:
> Is there a workflow to upgrade to 2.0?

Let's try to take this offline. Could you chat me, or create a discussion on github.com/toitlang/toit/discussions ?

askdesigners, 2022-10-21:
done https://github.com/toitlang/toit/discussions/1139