toitlang/toit #1047: Issues about porting ESP32-S3 to toit

donghengqaz, 2022-09-23:
Hi，

I am trying to port ESP32-S3 to toit, I mostly finish refactoring the Makefile, and compile final toit.bin. Now I meet an issues, system runs toit language binary failed, the log is:

```
Invoking primitive 0::15
LOAD_SMI_U8
LOAD_NULL
INVOKE_STATIC
AS_LOCAL
PRIMITIVE
Invoking primitive 0::15
THROW
fatal: "Threw exception before entering last-chance catch clause"
```

I add some log and find the _primitives is not initialized successfully, log is:

```
_primitives[INDEX_core(0)]=core_primitives(0x0)
_primitives[INDEX_timer(1)]=timer_primitives(0x0)
_primitives[INDEX_tcp(2)]=tcp_primitives(0x0)
_primitives[INDEX_udp(3)]=udp_primitives(0x0)
_primitives[INDEX_tls(4)]=tls_primitives(0x0)
_primitives[INDEX_esp32(5)]=esp32_primitives(0x0)
_primitives[INDEX_i2c(6)]=i2c_primitives(0x0)
_primitives[INDEX_i2s(7)]=i2s_primitives(0x0)
_primitives[INDEX_spi(8)]=spi_primitives(0x0)
_primitives[INDEX_spi_linux(9)]=spi_linux_primitives(0x0)
_primitives[INDEX_uart(10)]=uart_primitives(0x0)
_primitives[INDEX_rmt(11)]=rmt_primitives(0x0)
_primitives[INDEX_pcnt(12)]=pcnt_primitives(0x0)
_primitives[INDEX_crypto(13)]=crypto_primitives(0x0)
_primitives[INDEX_encoding(14)]=encoding_primitives(0x0)
_primitives[INDEX_font(15)]=font_primitives(0x0)
_primitives[INDEX_bitmap(16)]=bitmap_primitives(0x0)
_primitives[INDEX_events(17)]=events_primitives(0x0)
_primitives[INDEX_wifi(18)]=wifi_primitives(0x0)
_primitives[INDEX_ethernet(19)]=ethernet_primitives(0x0)
_primitives[INDEX_ble(20)]=ble_primitives(0x0)
_primitives[INDEX_dhcp(21)]=dhcp_primitives(0x3c068e94)
_primitives[INDEX_snapshot(22)]=snapshot_primitives(0x0)
_primitives[INDEX_image(23)]=image_primitives(0x0)
_primitives[INDEX_blob(24)]=blob_primitives(0x0)
_primitives[INDEX_gpio(25)]=gpio_primitives(0x0)
_primitives[INDEX_adc(26)]=adc_primitives(0x0)
_primitives[INDEX_dac(27)]=dac_primitives(0x0)
_primitives[INDEX_pwm(28)]=pwm_primitives(0x0)
_primitives[INDEX_touch(29)]=touch_primitives(0x0)
_primitives[INDEX_programs_registry(30)]=programs_registry_primitives(0x0)
_primitives[INDEX_flash(31)]=flash_primitives(0x0)
_primitives[INDEX_file(32)]=file_primitives(0x0)
_primitives[INDEX_pipe(33)]=pipe_primitives(0x0)
_primitives[INDEX_zlib(34)]=zlib_primitives(0x0)
_primitives[INDEX_subprocess(35)]=subprocess_primitives(0x0)
_primitives[INDEX_math(36)]=math_primitives(0x0)
_primitives[INDEX_x509(37)]=x509_primitives(0x0)
_primitives[INDEX_flash_kv(38)]=flash_kv_primitives(0x0)
_primitives[INDEX_debug(39)]=debug_primitives(0x0)
```
Could any one give me some help ?

kasperl, 2022-09-23:
Hi @donghengqaz! I can help, but maybe it is faster to take this on Discord (chat.toit.io)? 

We're (mis-)using weak linkage for the primitive tables, so maybe there is something fishy going on with that.

floitsch, 2022-09-23:
@mikkeldamsgaard reported that he had to migrate to the cmake-based build system to get his S3 working.
I wouldn't be surprised that the issue manifests itself this way.

donghengqaz, 2022-09-23:
Hi @kasperl @floitsch thanks for your reply. I have some progress, I modify cmake script like this:
```
target_link_libraries(${project_elf} "-Wl,--whole-archive -ltoit_vm -ltoit_image -Wl,--no-whole-archive")
```
And it works, S3 run hello.toit successfully. Do you have plan to add S3 recently ? If not, I would like to do this.

floitsch, 2022-09-23:
We already have some S3 boards on our tables, but won't have the time to work on them for the next few months.
If you could give it a try that would be awesome.

mikkeldamsgaard, 2022-09-26:
I did some investigations to determine the amount of work needed to migrate to idf cmake. It does not make sense to only do this for S3, so all supported "targets" should be included. The main pain point at the moment is to get the post processing with the @kasperl 's firmware script. The script commands themselves are easy, but convincing the idf build system to execute them is another matter. I ended up requiring calling a function from the toplevel CMakeList.txt. Let me work a little more on this, and I should be able to have something in a few days.
I do have a few S3 fixes in my code base (which is not a bit old, but most of the S3 related fixes should be current enough) here: https://github.com/mikkeldamsgaard/toit/tree/to-v4.4

mikkeldamsgaard, 2022-09-27:
And I converted that to a PR. @donghengqaz take a look at https://github.com/toitlang/toit/pull/1065

donghengqaz, 2022-09-28:
> And I converted that to a PR. @donghengqaz take a look at #1065

Hi @mikkeldamsgaard I see this PR and it's a good solution, and these days I also make a solution:https://github.com/toitlang/toit/pull/1068, I hope this can be a little helpful to you. By a way, I also find 2 issues:

1. compiling C3 will cause a error, and I create PR for this: https://github.com/toitware/esp-idf/pull/72, do you meet this issue ?
2. compiling with cmake based on esp-idf may cause more IRAM cost, then linking failed, for this I currently have to close this option: CONFIG_SPIRAM_CACHE_WORKAROUND, and try to find out the key, do you have some idea about this ?

mikkeldamsgaard, 2022-09-28:
Yes.
1. I did certainly encounter this issue and applied your patch in my checkout to work around it.
2. Indeed that was a main pain issue for me too. It seems very weird that cmake should increase the IRAM usage. But if you say it is due to the spiram cache workaround then that kind of makes sense as I believe the make based builds have had issues with spiram cache. Maybe @erikcorry can shed some light on this.

mikkeldamsgaard, 2022-09-28:
@donghengqaz If you do have insights into why cmake uses more IRAM, that would be helpful.

erikcorry, 2022-09-29:
We were pretty close to the IRAM size limit before.  We compile with -Os.  Could it be that the new build system compiles with -O2 or -O3 instead?

mikkeldamsgaard, 2022-09-29:
I did check and it should be -Os all over.
It is puzzling as the toit vm IRAM size also increases.

floitsch, 2022-09-29:
I looked at the commands that are used to compile the interpreter, and it looks like they are different now. I'm guessing that the old build system didn't take into account any compile-flag modifications we did in the src/CMakeLists.txt, whereas the new one does.
For example, we can see that there is now a `-O3` in the flags that wasn't there before. It is followed by a `-Os` so that one shouldn't matter, but other flags might. The interpreter is extra special since it has extra flags set in the CMakeLists.txt

old:
```
 -mlongcalls -DESP32 -DDEPLOY=1 -D__FREERTOS__=1 -Wno-sign-compare -DRAW=1 -fno-rtti  '-fdebug-prefix-map=/home/flo/code/opentoit=.' -Wall -ffunction-sections -fdata-sections -fno-exceptions  -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -Os -fno-rtti -DTOIT_DEPLOY -std=gnu++11 -O3 -fno-crossjumping -fno-tree-tail-merge  -o src/CMakeFiles/toit_core.dir/interpreter_run.cc.obj -c /home/flo/code/opentoit/src/interpreter_run.cc"
  
 -mlongcalls -DESP32 -DDEPLOY=1 -D__FREERTOS__=1 -Wno-sign-compare -DRAW=1 -fno-rtti  '-fdebug-prefix-map=/home/flo/code/opentoit=.' -Wall -ffunction-sections -fdata-sections -fno-exceptions  -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -Os -fno-rtti -DTOIT_DEPLOY -std=gnu++11 -o src/CMakeFiles/toit_core.dir/primitive_core.cc.obj -c /home/flo/code/opentoit/src/primitive_core.cc"
```

new:
```
 -mlongcalls -Wno-frame-address   '-fdebug-prefix-map=/home/flo/code/opentoit=.' -Wall -ffunction-sections -fdata-sections -fno-exceptions  -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -O3 -DNDEBUG -fno-rtti -DTOIT_DEPLOY -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Os -freorder-blocks -fmacro-prefix-map=/home/flo/code/opentoit/toolchains/esp32=. -fmacro-prefix-map=/home/flo/code/opentoit/third_party/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\\\"v3.2-dev-15749-g7f7cd0cec\\\" -DESP_PLATFORM -DNDEBUG -D_POSIX_READER_WRITER_LOCKS -std=gnu++11 -O3 -fno-crossjumping -fno-tree-tail-merge -o esp-idf/toit/toit/src/CMakeFiles/toit_core.dir/interpreter_run.cc.obj -c /home/flo/code/opentoit/src/interpreter_run.cc"
  
 -mlongcalls -Wno-frame-address   '-fdebug-prefix-map=/home/flo/code/opentoit=.' -Wall -ffunction-sections -fdata-sections -fno-exceptions  -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -O3 -DNDEBUG -fno-rtti -DTOIT_DEPLOY -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Os -freorder-blocks -fmacro-prefix-map=/home/flo/code/opentoit/toolchains/esp32=. -fmacro-prefix-map=/home/flo/code/opentoit/third_party/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\\\"v3.2-dev-15749-g7f7cd0cec\\\" -DESP_PLATFORM -DNDEBUG -D_POSIX_READER_WRITER_LOCKS -std=gnu++11 -o esp-idf/toit/toit/src/CMakeFiles/toit_core.dir/primitive_core.cc.obj -c /home/flo/code/opentoit/src/primitive_core.cc"
```

donghengqaz, 2022-09-30:
> @donghengqaz If you do have insights into why cmake uses more IRAM, that would be helpful.

I test and find the extra cost IRAM mostly is from libc, you can see that:

makefile:
```c
            Archive File DRAM .data .rtc.data DRAM .bss .rtc_noinit .rtc.text IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
                  libc.a         41         0         4           0         0        4389           0         4434       68971      6680                0          0       80081
```
cmake:
```c
            Archive File DRAM .data .rtc.data DRAM .bss .rtc_noinit .rtc.text IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
                  libc.a        365         0        38           0         0       11529           0        11932       79286      6714                0          0       97894
```
so I doubt cmake link some unused functions to IRAM, because libc functions' section are all ".text" not ".text.XXXX".

mikkeldamsgaard, 2022-09-30:
So, looking at esp32-spiram-rom-functions-c.lf and sdkconfig (SPIRAM_CACHE_LIBNUMPARSER_IN_IRAM), then for example lib_a-strtol should definitely be in IRAM. But make does not put it there and cmake does. This together with the  -mfix-esp32-psram-cache-strategy is what I think explains the difference. (And probably also why you never got spiram working on the ESP32??)
If you look at the generated sections.ld they are identical (weird).
A few options: 
1. Find the difference in libc functions and "disable" them in iram on cmake and keep the -mfix-esp32 disabled. (Will probably cause spiram to not work)
2. Not support ESP32 rev1, but only rev3+
3. Find something else to throw out of IRAM.

closed 2022-10-10 by donghengqaz