toitlang/toit #1596: crashing syntax highlighting

JUNGHUNKIM-7, 2023-04-30:
[image]

```
Thank you for amazing language.
1. Problem
- Syntax highlighting consistency looks abnormal.
```

kasperl, 2023-05-01:
Hi @JUNGHUNKIM-7. Thank you for the bug report. I think the issue might be related to the four space indentation style you're using here. Try using the recommended 2 space indentation and see if that helps.

https://docs.toit.io/language/style

kasperl, 2023-05-01:
The problem is that the syntax highlighter recognizes certain patterns based on their indentation (2 or 4 spaces). It would be great to fix this.

Here is an example:

https://github.com/toitware/ide-tools/blob/dbda69afa5a6fa1a5cccf50527da7bb680b9e861/vscode/syntaxes/toit.tmLanguage.yaml#L129

JUNGHUNKIM-7, 2023-05-01:
@kasperl 
Thank you for very detailed feedback! 
I should have checked the document carefully & never expected 2 space is official for such a very concised lang. if it has supported formatter in vscode, i could recognize it earlier.

Nice Project.

closed 2023-05-01 by JUNGHUNKIM-7