toitlang/toit #1660: BLE service.deploy crashes if a second service gets added

Informaticore, 2023-06-19:
If I create multiple services for an BLE peripheral I can always only add one service to it. Adding and deploying a second one always crashes with a core-panic.

```
s1 := peripheral.add_service uuid1
// set up s1 service
s1.deploy
s2 := peripheral.add_service uuid2
// setup s2 service
s2.deploy //<-- crash
```

Stacktrace:
```
******************************************************************************

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40130e5b  PS      : 0x00060e30  A0      : 0x80137c51  A1      : 0x3ffced80
A2      : 0x00000000  A3      : 0x3ffcedc0  A4      : 0x3ffc43a0  A5      : 0x0000cdcd
A6      : 0x00060e23  A7      : 0x00060e20  A8      : 0x8009aaa4  A9      : 0x3ffced70
A10     : 0x00000003  A11     : 0x00060e23  A12     : 0x00060e20  A13     : 0x3ffc49a8
A14     : 0x003fffff  A15     : 0x00000002  SAR     : 0x00000002  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40097c26  LEND    : 0x40097c31  LCOUNT  : 0x00000000

******************************************************************************
Decoding by `jag`, device has version <2.0.0-alpha.86>
******************************************************************************
Crash in native code:
Backtrace:0x40130e58:0x3ffced800x40137c4e:0x3ffceda0 0x40131f75:0x3ffcedc0 0x400d526f:0x3ffcedf0 0x40083f05:0x3ffcee20 0x400f91bc:0x3ffceea0 0x400f9462:0x3ffceef0 0x400f9491:0x3ffcef20 0x400ea4f2:0x3ffcef40 0x400ea519:0x3ffcef60
0x40130e58: ble_uuid_cmp + 0x0
0x40131f75: ble_gatts_start + 0xb5
0x400d526f: toit::primitive_deploy_service(toit::Process*, toit::Object**) + 0x167
0x40083f05: toit::Interpreter::run() + 0x254d
0x400f91bc: toit::Scheduler::run_process(toit::Locker&, toit::Process*, toit::SchedulerThread*) + 0x68
0x400f9462: toit::Scheduler::run(toit::SchedulerThread*) + 0x6a
0x400f9491: toit::SchedulerThread::entry() + 0x9
0x400ea4f2: toit::Thread::_boot() + 0x26
0x400ea519: toit::esp_thread_start(void*) + 0x5
******************************************************************************
```

Informaticore, 2023-06-19:
Okay then this sounds pretty much like my experience as well. When I wrote the test on the first try it did not crash but I thought I forgot to save and upload or something so I just tried again and then it crashed when deploying second service.

But I couldn't reproduce it not crashing so I stayed with my first thought. 
Maybe it is some memory issue then or even a ble stack problem deeper down.

Informaticore, 2023-06-19:
I also noticed something else but I am not sure if it is related: 
I am scanning for this service with the nRF Connect App and I connect with it as well. Sometimes I do see a GATT and a GAP Service (which afaik are mendatory) but 90% of the time I do not see them once I am connected. 
This is why I initially tried to add these Services by myself because I am not sure if toit is adding this services already or not.

kasperl, 2023-06-20:
I found a double free in the BLE implementation and what looks like a few memory leaks. I've attempted to address them here: https://github.com/toitlang/toit/pull/1662.

kasperl, 2023-06-21:
This looks like it is likely caused by https://github.com/apache/mynewt-nimble/issues/556 because we do not define all services in one go before calling `ble_gatts_start`.

commit 3842cf49 (2024-04-23): Start services in one go. (#2244)

Fixes #1660.
closed 2024-04-23 by floitsch