toitlang/toit #1694: Ethernet connection error makes device unreachable

Informaticore, 2023-07-10:
While using Artemis with Ethernet the device prints 
`E (60107) esp.emac: no mem for receive buffer` 
from time to time and sometimes multiple times. After that the Artemis `artemis.synchronize` process can not connect to its host anymore:

```
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
E (712677) esp.emac: no mem for receive buffer
E (712677) esp.emac: no mem for receive buffer
E (712677) esp.emac: no mem for receive buffer
E (712807) esp.emac: no mem for receive buffer
E (712807) esp.emac: no mem for receive buffer
E (713687) esp.emac: no mem for receive buffer
E (713687) esp.emac: no mem for receive buffer
E (714057) esp.emac: no mem for receive buffer
E (714807) esp.emac: no mem for receive buffer
[artemis.synchronize] WARN: connection to broker lost {error: DEADLINE_EXCEEDED}
[artemis.synchronize] INFO: disconnected
[artemis.scheduler] INFO: job stopped {job: synchronize}
[artemis.scheduler] INFO: job started {job: synchronize}
[artemis.synchronize] INFO: connecting
[artemis.synchronize] INFO: connected to network
[artemis.synchronize] WARN: connection to network lost {error: DEADLINE_EXCEEDED}
[artemis.synchronize] INFO: disconnected
```

After talking to @kasperl he found out that this might be an issue of ESP-IDF see: https://github.com/espressif/esp-idf/issues/9308

and has a possible Fix here: https://github.com/espressif/esp-idf/commit/6fff81d9700c6f615b711f7581329eea3566f32a

This issue appears to happen very easily on my device and not only the synchronize process dies but also my connections which means the devices become unavailable. This does not repairs itself as well, once it happened the device stays unconnected.

```
$ artemis version
v0.9.0

$ jag version                                                                                     
Version:         v1.10.0
SDK version:     v2.0.0-alpha.90     
Build date:      2023-07-06T15:43:49Z

$ toit version
+---------+------------+
| VERSION |    DATE    |
+---------+------------+
| v1.19.7 | 2022-03-07 |
+---------+------------+
```

Informaticore, 2023-08-09:
This seems still be an issue:

```
E (295612) esp.emac: no mem for receive buffer
E (295612) esp.emac: no mem for receive buffer
E (295612) esp.emac: no mem for receive buffer
E (295622) esp.emac: no mem for receive buffer
E (295622) esp.emac: no mem for receive buffer
E (295632) esp.emac: no mem for receive buffer
E (296402) esp.emac: no mem for receive buffer
E (296612) esp.emac: no mem for receive buffer
E (296622) esp.emac: no mem for receive buffer
[artemis.synchronize] WARN: connection to broker lost {error: DEADLINE_EXCEEDED}
[artemis.synchronize] INFO: disconnected
[artemis.scheduler] INFO: job stopped {job: synchronize}
[artemis.scheduler] INFO: job started {job: synchronize}
[artemis.synchronize] INFO: connecting
[artemis.synchronize] INFO: connected to network
[ble] DEBUG: Received BLE Payload #[0x65, 0x79, 0x4a, 0x68, 0x62, 0x47, 0x63, 0x69, 0x4f, 0x69, 0x4a, 0x49, 0x55, 0x7a, 0x49, 0x31, 0x4e, 0x69, 0x49, 0x
73, 0x49, 0x6e, 0x52, 0x35, 0x63, 0x43, 0x49, 0x36, 0x49, 0x6b, 0x70, 0x58, 0x56, 0x43, 0x4a, 0x39, 0x2e, 0x65, 0x79, 0x4a, 0x7a, 0x64, 0x57, 0x49, 0x69
, 0x4f, 0x69, 0x49, 0x78, 0x4d, ...]
[mqtt] DEBUG: try send unlock request to MQTT
[artemis.synchronize] WARN: connection to network lost {error: DEADLINE_EXCEEDED}
[artemis.synchronize] INFO: disconnected
[artemis.synchronize] INFO: connecting to broker failed - retrying
[artemis.synchronize] INFO: connecting
[artemis.synchronize] INFO: connected to network

```

It randomly happened at any time. 

```
artemis version
v0.9.1

jag version
Version:         v1.10.1
SDK version:     v2.0.0-alpha.93
Build date:      2023-08-08T08:40:28Z
```

Informaticore, 2023-08-09:
I was letting the code running for some time again. In the Log you can see that an error "DEADLINE_EXCEEDED" happened and then the device was disconnected. After a few attempts it stopped trying?! 
At the very end I simply unplugged and re-plugged the RJ45 Network cable and the broker connected again but it seems the scheduler stopped working now? More then 30min is gone and there is still no synchronisation happening.

```
[artemis.synchronize] INFO: synchronized
[ble] DEBUG: Received BLE Payload #[0x65, 0x79, 0x4a, 0x68, 0x62, 0x47, 0x63, 0x69, 0x4f, 0x69, 0x4a, 0x49, 0x55, 0x7a, 0x4
9, 0x31, 0x4e, 0x69, 0x49, 0x73, 0x49, 0x6e, 0x52, 0x35, 0x63, 0x43, 0x49, 0x36, 0x49, 0x6b, 0x70, 0x58, 0x56, 0x43, 0x4a,
0x39, 0x2e, 0x65, 0x79, 0x4a, 0x7a, 0x64, 0x57, 0x49, 0x69, 0x4f, 0x69, 0x49, 0x78, 0x4d, ...]
[mqtt] DEBUG: try send unlock request to MQTT
[mqtt] DEBUG: message received at sedisto/sbc73d3432d/unlock/response
[ble] DEBUG: Received BLE Payload #[0x65, 0x79, 0x4a, 0x68, 0x62, 0x47, 0x63, 0x69, 0x4f, 0x69, 0x4a, 0x49, 0x55, 0x7a, 0x4
9, 0x31, 0x4e, 0x69, 0x49, 0x73, 0x49, 0x6e, 0x52, 0x35, 0x63, 0x43, 0x49, 0x36, 0x49, 0x6b, 0x70, 0x58, 0x56, 0x43, 0x4a,
0x39, 0x2e, 0x65, 0x79, 0x4a, 0x7a, 0x64, 0x57, 0x49, 0x69, 0x4f, 0x69, 0x49, 0x78, 0x4d, ...]
[mqtt] DEBUG: try send unlock request to MQTT
[artemis.synchronize] INFO: synchronized
[mqtt] DEBUG: message received at sedisto/sbc73d3432d/unlock/response
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.scheduler] INFO: job started {job: ntp}
[artemis.ntp] INFO: synchronized {adjustment: 3.537405ms, time: 2023-08-09T15:31:43.838247}
[artemis.scheduler] INFO: job stopped {job: ntp}
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[artemis.synchronize] INFO: synchronized
[... 116 lines cut ...]
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Attempting to (re)connect
DEBUG: Connected to broker
DEBUG: Connection established
[ble] DEBUG: Received BLE Payload #[0x65, 0x79, 0x4a, 0x68, 0x62, 0x47, 0x63, 0x69, 0x4f, 0x69, 0x4a, 0x49, 0x55, 0x7a, 0x4
9, 0x31, 0x4e, 0x69, 0x49, 0x73, 0x49, 0x6e, 0x52, 0x35, 0x63, 0x43, 0x49, 0x36, 0x49, 0x6b, 0x70, 0x58, 0x56, 0x43, 0x4a,
0x39, 0x2e, 0x65, 0x79, 0x4a, 0x7a, 0x64, 0x57, 0x49, 0x69, 0x4f, 0x69, 0x49, 0x78, 0x4d, ...]
[mqtt] DEBUG: try send unlock request to MQTT
[mqtt] DEBUG: message received at sedisto/sbc73d3432d/unlock/response

```

kasperl, 2023-08-11:
As far as I can tell, the bug fix has been merged into ESP-IDF v5.0.1. We've been looking at upgrading to v5.0 for a little while: https://github.com/toitlang/toit/pull/1583. I'll talk to @erikcorry and try to figure out where we are.

kasperl, 2023-08-16:
@Informaticore We have landed the first part of the upgrade to ESP-IDF v5.0 on master just now. We'll be updating to the latest v5.0.x bits over the next few days.

kasperl, 2023-08-21:
The fix has landed in SDK v2.0.0-alpha.96.

closed 2023-09-09 by kasperl