toitlang/toit #1806: Espnow tutorials needed

nne998, 2023-09-13:
Hi,
Is there any espnow tutorials for esp32c3 chip ?

Thank you

floitsch, 2023-09-13:
We don't have a tutorial yet, but there is an example here: https://github.com/toitlang/toit/blob/master/examples/espnow.toit

Note that Toit doesn't support running espnow at the same time as WiFi.

If you are using Jaguar, you have to disable Jaguar, so it doesn't start up the WiFi while the espnow program is running.

```
jag container install -D jag.disabled -D jag.timeout=2m  espnow examples/espnow.toit
```

This would disable Jaguar for 2 minutes while the espnow example is running.

kasperl, 2024-11-16:
@kidandcat Artemis allows you to stop the networking while your app runs, so that way you can programatically insist that Artemis doens't interfere with your WiFi / espnow usage.

https://pkg.toit.io/github.com/toitware/toit-artemis@0.10.0/docs/artemis/library-summary#run(3%2C1%2C0%2Coffline%2Ctimeout)