toitlang/toit #2149: "an instance with class-id 10"

igouy, 2024-03-17:
Poking-around, trying to trace through code snippets, `print`  shows "class-id 10".

Is there somewhere (even a table in the docs) where we could lookup what class that is, and then go search for more info about that class in the docs?

igouy, 2024-03-18:
Just break something:

`Class 'Stream' does not have any method 'plop'.`

closed 2024-03-18 by igouy

floitsch, 2024-03-18:
The default `stringify` method of classes doesn't include their name. This is to save space on smaller devices.
However, an unstripped snapshot does. If you first compile to a snapshot, you can then list all classes with their ids by running `toitp --classes program.snapshot`, where `toitp` is located in the tools folder of the SDK.

The `Stream` class is from the `host` package: https://pkg.toit.io/github.com/toitlang/pkg-host@1.14.1/docs/host/file/class-Stream