toitlang/toit #2173: fwiw my new mistakes

igouy, 2024-03-21:
Perhaps because I'm transliterating programs written in other languages into Toit, the mistakes surprise me a little.

In the original program `/` converted to float division but in the transliterated Toit program was interpreted as int division so the program mysteriously produced unexpected output.

Indented a statement 2-spaces too many which put the statement inside a loop so the program mysteriously produced unexpected output.

floitsch, 2024-03-22:
For the division, Toit is in "good" company. Lots of languages have int / int -> int, but float / int->float and int / float -> float.

For the indentation: you get used to it :), and eventually appreciate that you don't need to declare your indent twice (with delimiters and indentation).

Fwiw, as a human, you also run in the opposite issue, where code (in a delimited language) was misread because of the indentation. See https://developers.redhat.com/blog/2016/02/26/gcc-6-wmisleading-indentation-vs-goto-fail. That said: in delimited languages the compiler can warn you. In Toit indentation is the only way to declare indent.

I do think that transliterating makes this worse, as you don't necessarily think about the purpose of the code.

floitsch, 2024-03-22:
And thanks for the feedback.

igouy, 2024-07-12:
fyi 

https://www.reddit.com/r/programming/comments/1e11kly/toit_is_a_modern_highlevel_language_designed/

igouy, 2024-07-16:
> Toit runs 10-20x faster than MicroPython

_fwiw_ I see 2x on linux

$ hyperfine " /opt/src/micropython/micropython fannkuchredux.micropython-6.micropython 11"
Benchmark 1:  /opt/src/micropython/micropython fannkuchredux.micropython-6.micropython 11
  Time (mean ± σ):     152.510 s ±  3.891 s    [User: 152.478 s, System: 0.021 s]
  Range (min … max):   146.851 s … 158.057 s    10 runs

$ hyperfine "/opt/src/micropython/micropython  -X emit=native fannkuchredux.micropython-6.micropython 11"
Benchmark 1: /opt/src/micropython/micropython  -X emit=native fannkuchredux.micropython-6.micropython 11
  Time (mean ± σ):     121.711 s ±  3.740 s    [User: 121.656 s, System: 0.027 s]
  Range (min … max):   118.140 s … 131.472 s    10 runs

$ hyperfine "./fannkuchredux.toit_run 11"
Benchmark 1: ./fannkuchredux.toit_run 11
  Time (mean ± σ):     95.785 s ±  3.115 s    [User: 95.668 s, System: 0.096 s]
  Range (min … max):   91.944 s … 99.143 s    10 runs

floitsch, 2024-07-16:
> > Toit runs 10-20x faster than MicroPython
> 
> _fwiw_ I see 2x on linux

Fannkuch is a microbenchmark that only exercises a small part of the interpreter. Depending on the implementation there are typically almost no classes and virtually no method calls. Deltablue and Richards are benchmarks that are much more representative of real-word applications.

In other words: some operations in Toit run roughly at the same speed as MicroPython. Others are significantly faster. For typical applications, with classes and method calls, the difference becomes more pronounced. Microbenchmarks typically only test loop operations and number-crunching which Toit isn't as optimized for.