toitlang/toit #2184: Toit compilation failed on Ubuntu 22.04.

HKPhysicist, 2024-04-01:
Hello Friend,
I can compile Toit release 1.6 but the newest v2.0.0-alpha.143 fails to compile.  Here are my whole process.

sudo apt install build-essential cmake ninja-build golang-1.21 python-is-python3 glibc-tools
cd /usr/local/src/
sudo git clone https://github.com/toitlang/toit.git
cd toit
sudo git submodule update --init --recursive
sudo fakeroot make all

Then, it showed errors:
===============
make rebuild-cmake
make[1]: Entering directory '/mnt/WD-Gold/local/src/toit'
mkdir -p build/host
(cd build/host && cmake /mnt/WD-Gold/local/src/toit -G Ninja -DTOITC=/mnt/WD-Gold/local/src/toit/build/host/sdk/bin/toit.compile -DTOITPKG=/mnt/WD-Gold/local/src/toit/build/host/sdk/bin/toit.pkg -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE=/mnt/WD-Gold/local/src/toit/toolchains/host.cmake --no-warn-unused-cli)
Not searching for unused variables given on the command line.
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at third_party/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:21 (cmake_minimum_required):
  The OLD behavior for policy CMP0076 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/WD-Gold/local/src/toit/build/host
make[1]: Leaving directory '/mnt/WD-Gold/local/src/toit'
(cd build/host && ninja build_tools)
[1/494] Generating sdk/bin/toit.lsp
FAILED: sdk/bin/toit.lsp /mnt/WD-Gold/local/src/toit/build/host/sdk/bin/toit.lsp 
cd /mnt/WD-Gold/local/src/toit/tools/toitlsp && /usr/bin/cmake -E env GODEBUG=netdns=go GOOS= GOARCH= go build -ldflags \ -X\ main.date=2024-04-01T11:01:25Z\ -X\ main.version=v2.0.0-alpha.143.5+bee9cd03 -tags netgo\ osusergo -o /mnt/WD-Gold/local/src/toit/build/host/sdk/bin/toit.lsp
No such file or directory
[2/494] Generating sdk/bin/toit.pkg
FAILED: sdk/bin/toit.pkg /mnt/WD-Gold/local/src/toit/build/host/sdk/bin/toit.pkg 
cd /mnt/WD-Gold/local/src/toit/tools/tpkg && /usr/bin/cmake -E env GODEBUG=netdns=go GOOS= GOARCH= go build -ldflags \ -X\ main.date=2024-04-01T11:01:25Z\ -X\ main.version=v2.0.0-alpha.143.5+bee9cd03 -tags netgo\ osusergo -o /mnt/WD-Gold/local/src/toit/build/host/sdk/bin/toit.pkg
No such file or directory
[6/494] Building CXX object src/CMakeFiles/toit_core.dir/encoder.cc.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:148: host-tools] Error 1

kasperl, 2024-04-01:
I think you're missing a go toolchain? 

https://github.com/toitlang/toit?tab=readme-ov-file#linux-and-mac

HKPhysicist, 2024-04-01:
> I think you're missing a go toolchain?
> 
> https://github.com/toitlang/toit?tab=readme-ov-file#linux-and-mac
> 
> I could be wrong though, because you seem to have installed that using `apt install`. Maybe there is something fishy with the fakeroot handling.

Yes, the default golang v1.18 executables are under /usr/bin/ but executables of golang v1.21 are not there.  I created links under /usr/bin/ point to golang v1.21 and compilation is successful.

However, the drivers of cellular modems are not there any more.  Are they still supported?

kasperl, 2024-04-01:
@HKPhysicist The drivers have migrated to a separate package: https://pkg.toit.io/package/github.com%2Ftoitware%2Fcellular.

HKPhysicist, 2024-04-01:
WOW!
Your Toitware + GPSd ublox tool will become a complete solution for ublox chip on Linux board!

Thank you very much!

floitsch, 2024-09-23:
I think this issue was resolved.

closed 2024-09-23 by floitsch