toitlang/toit #2254: Does custom class use less memory than List?

igouy, 2024-04-24:
No doubt I'm not seeing my mistake. No doubt the List program is wrongly allocating more children.

[binary trees using a Tree class](https://benchmarksgame-team.pages.debian.net/benchmarksgame/program/binarytrees-toit-1.html) --- 312,704 

[binary trees using List](https://benchmarksgame-team.pages.debian.net/benchmarksgame/program/binarytrees-toit-2.html) --- 724,864 	

Is this plausible?

floitsch, 2024-04-24:
Yes. This sounds plausible.

An object of a custom class is very efficient. It uses only 1 word (32 bits) as header, and then a word for each field.

A list, on the other hand, needs a header, a length, and a backing store where it stores the actual elements. For efficiency reasons, the backing store is furthermore often bigger than it needs to be, so that adding new elements doesn't always require to allocate a new backing store. If it only had the minimal required space, adding a new object would always require to copy everything into a new backing store, leading to quadratic performance behavior.

closed 2024-04-24 by floitsch

igouy, 2024-04-24:
Thanks. 

I incremented a global in the constructor and on the paths with [] both showed the same 613766494.