toitlang/toit #2439: Type-inference tests fail with inheritance test

floitsch, 2024-07-19:
I get the following error:
```
1/2 Test  #738: tests/negative/inheritance-test.toit ...   Passed    0.07 sec
1137: fatal: didn't expect 24 at 3238
```

floitsch, 2024-07-19:
The test fails here: https://github.com/toitlang/toit/blob/a8a5a92a23abbc9db6cd09df7959f1700e1c9237/tests/toitdoc/inheritance.toit#L46

Run with:
```
git checkout type-inference-test-failure
rm -rf build/host--ctp
LOCAL_CXXFLAGS="-O0" make HOST=host-ctp TOOLCHAIN=host TOIT_CHECK_PROPAGATED_TYPES=1 sdk
ctest --test-dir build/host-ctp --verbose -C slow -R toitdoc/inheritance-test
```

The `LOCAL_CXXFLAGS="-O0"` is optional.

floitsch, 2024-07-19:
Once fixed please change the code (master/PR) back to using a block.

kasperl, 2024-10-07:
This smaller examples illustrates the problem:

```
main:
  x := null
  b := : if x: print "hello"
  [ 1 ].do: x = it
  b.call
```

The assignment to `x` from within the block passed to `do` isn't handled correctly for some reason.

kasperl, 2024-10-22:
Addressed by #2597.

commit 4b4770ac (2024-11-08): Fix #2439 (#2614)
closed 2024-11-08 by kasperl