toitlang/toit #2493: assert: how-to ?

igouy, 2024-08-05:
```
richarts.toit

  assert: scheduler.queue-count == EXPECTED-QUEUE-COUNT
  assert: scheduler.hold-count == EXPECTED-HOLD-COUNT
```

I haven't been able to figure-out how to trigger the assert or where to see what happened.

floitsch, 2024-08-06:
It's currently a bit difficult to enable asserts.
You need to run the compiler by hand with `-Xenable-asserts`. That's clearly not user-friendly.

I will prioritize making this easier. Probably `-O0` will have it on by default.

floitsch, 2024-08-06:
I just uploaded a PR that makes `-O0` and `-O1` (the default) enable asserts.
There is also a flag to set enable-assert explicitly. We will expose it in Jaguar once a new SDK with this change has been released.

igouy, 2024-08-06:
> You need to run the compiler by hand with `-Xenable-asserts`. That's clearly not user-friendly.

I can do that !

Is there a way for me to discover the available compiler flags?

floitsch, 2024-08-08:
> > You need to run the compiler by hand with `-Xenable-asserts`. That's clearly not user-friendly.
> 
> I can do that !
> 
> Is there a way for me to discover the available compiler flags?

These compiler flags are mostly for testing and internal use, but you can find them here:
https://github.com/toitlang/toit/blob/master/src/flags.h#L45

commit d00d5ce2 (2024-08-06): Enable asserts by default for -O0 and -O1 and add flag.

Partially fixes #2493.
We still need to expose it in Jaguar.
PR #2494 (2024-09-20): Enable asserts by default for -O0 and -O1 and add flag.
Partially fixes #2493.
We still need to expose it in Jaguar.
commit 90a74b87 (2024-09-20): Enable asserts by default for -O0 and -O1 and add flag. (#2494)

Partially fixes #2493.
We still need to expose it in Jaguar.

---------
closed 2024-09-20 by floitsch