toitlang/toit #2512: BLE Example scan duration never times out on MacOS with Toit SDK v2.0.0

z3ugma, 2024-08-27:
Running the example ble example from https://github.com/toitlang/toit/blob/master/examples/ble/scan.toit just results in a hanging process that never times out on MacOS. 

The BLE Central scan never times out, as though it's not respecting the scan duration param? 

```
sw_vers
ProductName:            macOS
ProductVersion:         14.5
BuildVersion:           23F79

jag version
Version:         v1.39.0
SDK version:     v2.0.0-alpha.159
Build date:      2024-08-27T19:37:04Z
```

I have noticed this specifically with the v2.0.0 alpha SDK but it was working fine in the past.

z3ugma, 2024-08-27:
Also doesn't work on my setup with Jaguar v1.36.3 and v2.0.0-alpha.149 

Downloading Toit SDK from https://github.com/toitlang/toit/releases/download/v2.0.0-alpha.149/toit-macos.tar.gz ...
52.50 MiB / 52.50 MiB [-------------------------------------------------------------------------------] 100.00% 9.76 MiB p/s
Successfully installed Toit SDK into /Users/me/.cache/jaguar/sdk
Downloading Jaguar assets from https://github.com/toitlang/jaguar/releases/download/v1.36.3/assets.tar.gz ...
Successfully installed Jaguar assets into /Users/me/.cache/jaguar/assets
145.75 KiB / 145.75 KiB [------------------------------------------------------------------------------------] 100.00% ? p/s
Successfully setup Jaguar v1.36.3 with Toit SDK v2.0.0-alpha.149.

floitsch, 2024-08-28:
I did some changes to the BLE stack with alpha.147. I guess I broke the timeout then.

I don't have any macos machine accessible, but will have a look and try to find the issue by just looking at the code.
If I don't find anything @kasperl might need to help.

z3ugma, 2024-08-28:
Trying it on a different workstation, I notice that I get a log line: 
`cm started`  that I don't get on the problem workstation
that's 
```
Version:         v1.33.0
SDK version:     v2.0.0-alpha.145
Build date:      2024-04-22T20:36:07Z

sw_vers
ProductName:            macOS
ProductVersion:         13.6.7
BuildVersion:           22G720
```

floitsch, 2024-08-29:
The 'cm started' comes from 'centralManagerDidUpdateState':
https://github.com/toitlang/toit/blob/master/src/resources/ble_darwin.mm#L502

I would expect to see this line as soon as the adapter is powered on.

Could you verify that it's really the scan function that fails, and not already something earlier? Basically, just add a `print` in front of the `central.scan`.

floitsch, 2024-08-29:
> Trying it on a different workstation, I notice that I get a log line: `cm started` that I don't get on the problem workstation that's
>
> ```
> Version:         v1.33.0
> SDK version:     v2.0.0-alpha.145 [...]

On this other machine does the Bluetooth work?

If yes, then maybe the macOS version (13 vs 14) is the reason.

PR #2513 (2024-08-29): Fix BLE on macos.
Fixes #2512.
commit 73da4257 (2024-08-29): Fix BLE on macos. (#2513)

Fixes #2512.

---------
closed 2024-08-29 by floitsch