toitlang/toit #2604: Is there an io.Reader stdin read-line size limit?

igouy, 2024-11-01:
See the attached program and example data.

`./test < data.txt`

data.txt size 130984 and smaller print non-zero size

data.txt size 130995 and larger print zero size

[test.txt](https://github.com/user-attachments/files/17602953/test.txt)

[data.txt](https://github.com/user-attachments/files/17602955/data.txt)

floitsch, 2024-11-05:
Could you provide a data.txt that fails for you?
I just tried with an 11MB file and got a non-zero output.

igouy, 2024-11-07:
[kin50000.txt](https://github.com/user-attachments/files/17666624/kin50000.txt)

floitsch, 2024-11-08:
The test.toit program (attached as `test.txt` here) has the following contents:
```
import host.pipe
import io
    
seq_lines -> List :  
  lines := []
  stdin := pipe.stdin.read 
  if stdin : 
    in ::= io.Reader stdin
    while line := in.read-line :    
      if line.starts-with ">THREE" : break   
    while line := in.read-line :  
      if line.starts-with ">" : break    
      lines.add line  
  return lines
    
main args:
  print seq_lines.size
```

Note that it starts by `stdin := pipe.stdin.read`
This only reads a chunk from stdin. There might be more.

I suggest to rewrite the function as follows:
```
import host.pipe

seq-lines -> List:
  lines := []
  in := pipe.stdin.in
  while line := in.read-line:
    if line.starts-with ">THREE": break 
  while line := in.read-line:
    if line.starts-with ">": break
    lines.add line
  return lines

main args:
  print seq-lines.size
```

The `pipe.stdin` already provides a reader as `.in` so no need to create a new one. After that things should work.

Note: the type of `pipe.stdin` is currently `any` which makes it hard to find the `in` since there is no auto-completion.  We are aware of this issue and have plans to fix it. (Unfortunately not as trivial as it sounds).

closed 2024-11-08 by floitsch

floitsch, 2024-11-11:
The mistake was to assume that `pipe.stdin.read` would read the complete input.

From Toit's side we can make things better by giving `pipe.stdin` a type, as that would give auto-completion which would have (maybe) led you to the `in` getter which is already a reader.
We have it on our list to rewrite/refactor the `host` package.