toitlang/toit #2696: wake up by ulp from deep sleep?

JunfengJia, 2025-01-17:
My application is power-hungry, so I’m looking for a way to put the ESP32 into deep sleep and wake it up using a pin or some other logic running in the ULP. Any ideas?

kasperl, 2025-01-17:
@JunfengJia With a little bit of luck, you may not need the ULP to trigger this. Here is an example of using a GPIO pin trigger for waking up from deep sleep: https://github.com/toitlang/toit/blob/master/examples/triggers/gpio.toit.

JunfengJia, 2025-01-17:
Thanks a bunch, @kasperl! The GPIO works for me in some cases, like when the user presses a button. I’ll do more testing for other scenarios. I also created a small ulp + c program that works for the basic idea, but I’m not sure if it can also work with ULP+toit.  ULP, check condition, if true, wakeup esp32, then C code do some work, then deep sleep, and ULP checking again. Do you think there’s a way to do it in ULP + toit?

floitsch, 2025-01-20:
We haven't implemented support for ULP yet.
There seems to be some interest in it, so I will try to bump its priority. Will probably still take a few weeks before I get to it.

 Once implemented, your use case should be possible.

JunfengJia, 2025-01-20:
Hey @floitsch, thanks for getting back to me! For now, I’ll use sleep/check as a solution. Thanks again for working on it!