toitlang/toit #2756: STACK_OVERFLOW <sdk>/i2c.toit:333 since v2.0.0-alpha.178

addshore, 2025-03-31:
When updating to toit v2.0.0-alpha.178 and jag [v1.50.2](https://github.com/toitlang/jaguar/releases/tag/v1.50.2) running code that previously worked:

```
******************************************************************************
EXCEPTION error.
STACK_OVERFLOW
  0: Device.read-address.<block> <sdk>/i2c.toit:333:39
  1: Device.with-failure-handling_ <sdk>/i2c.toit:199:12
  2: Device.read-address       <sdk>/i2c.toit:340:12
  3: Device.read-address       <sdk>/i2c.toit:333:12
  4: Writer.try-write_         <pkg:toit-lightbug>/devices/i2c.toit:112:27
  5: Writer.try-write          <sdk>/io/writer.toit:110:16
  6: Writer.write              <sdk>/io/writer.toit:51:14
  7: Comms.sendSwitching_      <pkg:toit-lightbug>/services/comms/comms.toit:294:19
  8: Comms.send                <pkg:toit-lightbug>/services/comms/comms.toit:273:5
  9: Comms.sendOpen_           <pkg:toit-lightbug>/services/comms/comms.toit:77:13
 10: Comms.start_              <pkg:toit-lightbug>/services/comms/comms.toit:70:7
 11: Comms                     <pkg:toit-lightbug>/services/comms/comms.toit:58:5
 12: main                      ...........
******************************************************************************
```

kasperl, 2025-03-31:
That is not a lot of frames for a stack overflow, but the code does look broken on failures:

```
  /**
  Reads $size bytes from the given $address.
  */
  read-address address/ByteArray size/int -> ByteArray:
    return read-address address size: throw it

  /**
  Variant of $(read-address address size).
  Calls the $failure block if the operation fails.
  */
  read-address address/ByteArray size/int [failure] -> ByteArray:
    return with-failure-handling_ --on-failure=failure:
      read-address address size
```

These two look mutually recursive to me.

kasperl, 2025-03-31:
I think maybe this would be better, but I'm a bit of an i2c noob:

```
  /**
  Reads $size bytes from the given $address.
  */
  read-address address/ByteArray size/int -> ByteArray:
    return write-read address size
```

kasperl, 2025-03-31:
> So, if I wanted to test your proposed fix, whats the best way to go about it for me? I'd have to build by own esp32c6 envelope, right? with the adjusted toit code?

Already mentioned on Discord, but for the record: The code is just library code, so you should be able to just patch the SDK library source in place - and re-run your application.

PR #2757 (2025-03-31): Fix I2c registers.
Fixes #2756.
commit 33b04be1 (2025-03-31): Fix I2c registers. (#2757)

Fixes #2756.
closed 2025-03-31 by floitsch