toitlang/toit #2758: Try/finally not counted as definite assignment

floitsch, 2025-03-31:
The following program yields an error:
```
class A:
  foo/int

  constructor:
    try:
      foo = 499
    finally:
      print "done"
```

```
ᐅ toit analyze /tmp/t.toit
/tmp/t.toit:4:3: error: Field 'foo' not initialized on all paths in constructor
  constructor:
  ^~~~~~~~~~~
```

kasperl, 2025-03-31:
I guess you have to be a little bit careful with ways to get out of the `finally` block without unwinding, thus continuing to construct the object without necessarily having run through the entire `try` block.

```
class A:
  foo/int
  constructor:
    while true:
      woops := : break
      try:
        [0,1,2][3]
        foo = 87
      finally:
        woops.call
```

floitsch, 2025-03-31:
Agreed.
For field-initialization that's fortunately not a problem. If we exit the constructor without an exception, then we must have gone through all statements in the try-block. And the finally doesn't have a way to change an exception into a "normal" return, as that would change the constructor into a factory.

kasperl, 2025-04-01:
Have you tried to run my example? It does not throw an exception, constructs the objects, and does not set the field to 87.

floitsch, 2025-04-01:
> Have you tried to run my example?
Not correctly :)